kiddin9/Kwrt #753: IPv6中继模式失败

xw-yin, 2022-06-16:
中继模式配置完成之后设备可以获取IPv6地址，但是IPv6test测试失败，也无法访问IPv6网站。奇怪的是我把配置备份恢复到其他固件居然成功了就很奇怪。

kiddin9, 2022-06-16:
关闭梯子再测试看看.

xw-yin, 2022-06-16:
关了还是不行

xw-yin, 2022-06-16:
还有一个bug就是文件管理的文件列表，这个版本显示不出来，但是上个版本就可以。

xw-yin, 2022-06-16:
[image]
我说的是这里，我记得上个版本这里是显示文件夹的列表的。

kiddin9, 2022-06-16:
是这里,我的正常显示,

xw-yin, 2022-06-16:
[image] 
我甚至用手机浏览器也试了还是没有，这是我的设备和固件信息。

xuebkgithub, 2022-06-16:
ipv6 nat 模式下.需要自定义 路由表.你自定义了嘛

xw-yin, 2022-06-16:
不是nat 是relay

xuebkgithub, 2022-06-17:
你用中继的时候.dns获取到了吗?  另外一定要关闭梯子和各种代理
你贴下.你的dhcp 和 naetwork

xw-yin, 2022-06-17:
[image]
[image]

xw-yin, 2022-06-17:
config dnsmasq
	option domainneeded '1'
	option localise_queries '1'
	option local '/lan/'
	option domain 'lan'
	option expandhosts '1'
	option authoritative '1'
	option readethers '1'
	option leasefile '/tmp/dhcp.leases'
	option ednspacket_max '1232'
	option port '53'
	option dns_redirect '1'
	option allservers '1'
	option min_ttl '3600'
	option cachesize '8192'
	option dnsforwardmax '8192'
	option localservice '0'
	option rebind_protection '0'
	option noresolv '1'
	list server '127.0.0.1#7874'

config dhcp 'lan'
	option interface 'lan'
	option start '100'
	option limit '150'
	option leasetime '12h'
	option dhcpv4 'server'
	option force '1'
	option ra 'relay'
	option dhcpv6 'relay'
	option ndp 'relay'

config dhcp 'wan'
	option interface 'wan'
	option ignore '1'
	option ra 'relay'
	option dhcpv6 'relay'
	option ndp 'relay'

config odhcpd 'odhcpd'
	option maindhcp '0'
	option leasefile '/tmp/hosts/odhcpd'
	option leasetrigger '/usr/sbin/odhcpd-update'
	option loglevel '4'

config dhcp 'wan6'
	option interface 'wan6'
	option ignore '1'
	option ra 'relay'
	option dhcpv6 'relay'
	option ndp 'relay'
	option master '1'

xw-yin, 2022-06-17:
config interface 'loopback'
	option device 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'

config globals 'globals'
	option packet_steering '1'

config device
	option name 'br-lan'
	option type 'bridge'
	list ports 'eth1'
	list ports 'eth2'
	list ports 'eth3'

config interface 'lan'
	option device 'br-lan'
	option proto 'static'
	option netmask '255.255.255.0'
	option ip6assign '60'
	option ipaddr '192.168.2.1'

config interface 'wan'
	option device 'eth0'
	option proto 'dhcp'
	option ipv6 'auto'

config interface 'wan6'
	option proto 'dhcpv6'
	option device 'eth0'
	option reqaddress 'try'
	option reqprefix 'auto'

xuebkgithub, 2022-06-17:
我看你的电脑截图里面.是已经获取到ipv6了啊. 地址.和dns都正常啊.
你用 traceroute6 6.ipw.cn 看下你的路由可以行走吗?
下图是我正常的
[image]

xw-yin, 2022-06-17:
[image]

xuebkgithub, 2022-06-17:
你关闭翻墙.(增强模式.设置系统代理.).在关闭ipv4只保留ipv6 自动获取.  这样的话.你获取到ipv6地址了吗.

xuebkgithub, 2022-06-17:
你删除 option ip6assign '60'

xuebkgithub, 2022-06-17:
你ping6 你的网关, 路由器.和路由器dhcp ipv6获取到的自己.看下通不通

xw-yin, 2022-06-17:
我路由搞了内网穿透  你登录看下
xw-yin.icu:8003
密码 123456

xw-yin, 2022-06-17:
ping6 路由地址是可以的

xw-yin, 2022-06-17:
我换个固件试试吧

xuebkgithub, 2022-06-17:
刚到家.还没来得及回你

xw-yin, 2022-06-17:
刚刷的别人的5.15内核的固件也不行，但是刷的5.10的固件就可以

xuebkgithub, 2022-06-17:
刚刚把你的路由器 去掉了option ip6assign '60'  并且重启了下.你在看下

xw-yin, 2022-06-17:
[image]

xuebkgithub, 2022-06-17:
你在看下呢?

xw-yin, 2022-06-17:
这次直接连公网ipv6地址也没了

xuebkgithub, 2022-06-17:
又调整了下.再不行的话.你就换成 nat6把

xuebkgithub, 2022-06-17:
你电脑现在 分配到ipv6地址了吗?

xw-yin, 2022-06-17:
还是不行 刚看了下openwrt的issue也有同样问题，没办法了，辛苦了，谢谢

xw-yin, 2022-06-17:
https://github.com/openwrt/openwrt/issues/9881

xuebkgithub, 2022-06-17:
要不你换成nat6模式?

xw-yin, 2022-06-17:
算了不搞了，内网穿透也够用了，什么时候搞个国内服务器体验应该比IPv6还好，毕竟国内大多都没有公网IPv6环境也访问不了。

xw-yin, 2022-06-17:
zerotier 设备还要装客户端，手机上还要用vpn模式运行太麻烦了，还是frp方便。

xw-yin, 2022-06-17:
[image]
[image]
[image]
直接把备份恢复到这个固件就有ipv6公网了，应该是openwrt源码有什么变动。

dmlt, 2022-06-28:
一样的情况，电脑分配到了ipv6公网ip，能解析ipv6 地址，就是ping6不通。

xuebkgithub, 2022-07-04:
转发.搞一下
#!/bin/sh /etc/rc.common
# Example script
# Copyright (C) 2007 OpenWrt.org

START=99
STOP=15
interface_public="wan6"
res=`ip -6 route | grep "default from"`
gateway=`echo $res | awk '{print $5}'`
interface=`echo $res | awk '{print $7}'`
start() {
    ip -6 r add default via $gateway dev $interface
    ip6tables -t nat -A POSTROUTING -o $interface -m comment --comment "v6NAT" -j MASQUERADE
}

stop() {
    ip6tables -t nat -D POSTROUTING -o $interface -m comment --comment "v6NAT" -j MASQUERADE
    ip -6 r del default via $gateway dev $interface
}

closed 2023-01-07 by xw-yin