kiddin9/Kwrt #760: R5S作为wireguard客户端无法创建网络接口，固件版本 06.19.2022

chainone, 2022-06-19:
看了下，几个关键组件都已经安装了，比如macvlan, 重启也重启了好几遍，都不奏效，在“WIREGUARD状态”界面始终显示没有接口

Sun Jun 19 10:42:36 2022 daemon.notice netifd: Interface 'WG0' is setting up now
Sun Jun 19 10:42:36 2022 daemon.notice netifd: WG0 (15203): RTNETLINK answers: Not supported
Sun Jun 19 10:42:37 2022 daemon.notice netifd: WG0 (15203): Unable to modify interface: Protocol not supported
Sun Jun 19 10:42:42 2022 daemon.notice netifd: Interface 'WG0' is now down
Sun Jun 19 10:42:42 2022 daemon.notice netifd: Interface 'WG0' is setting up now
Sun Jun 19 10:42:42 2022 daemon.notice netifd: WG0 (15307): RTNETLINK answers: Not supported
Sun Jun 19 10:42:43 2022 daemon.notice netifd: WG0 (15307): Unable to modify interface: Protocol not supported
Sun Jun 19 10:42:48 2022 daemon.notice netifd: Interface 'WG0' is now down
Sun Jun 19 10:42:48 2022 daemon.notice netifd: Interface 'WG0' is setting up now
Sun Jun 19 10:42:48 2022 daemon.notice netifd: WG0 (15411): RTNETLINK answers: Not supported
Sun Jun 19 10:42:49 2022 daemon.notice netifd: WG0 (15411): Unable to modify interface: Protocol not supported
Sun Jun 19 10:42:54 2022 daemon.notice netifd: Interface 'WG0' is now down
Sun Jun 19 10:42:54 2022 daemon.notice netifd: Interface 'WG0' is setting up now
Sun Jun 19 10:42:54 2022 daemon.notice netifd: WG0 (15515): RTNETLINK answers: Not supported
Sun Jun 19 10:42:55 2022 daemon.notice netifd: WG0 (15515): Unable to modify interface: Protocol not supported

chainone, 2022-06-19:
分支: 22.03.0-rc4
内核: 5.10.120

chainone, 2022-06-20:
求解决

chainone, 2022-06-23:
重新用了R5S的默认固件也不行…应该是通性问题了，是不是什么内核模块没装？

chainone, 2022-07-13:
最新的0710版此问题还是存在

chainone, 2022-07-13:
@kiddin9 请帮忙看看为啥wireguard在R5S上跑不起来，谢谢！

chainone, 2022-07-26:
最新版0724试了还是不行
@kiddin9 能否帮忙明确下，如果R5S暂不打算支持wireguard, 那我就寻找别的方式了，不再出一个版本试一下了。

kiddin9, 2022-07-26:
目前没办法, 内核用的是R5S官方已编译好的,他的内核不支持wireguard模块,

clhlc, 2024-03-12:
最新r2s 也不行