kiddin9/Kwrt #762: 6.18更新版本 x86概览里面不显示cpu温度了

656567543, 2022-06-19:
6.18更新版之后用定制固件 x86概览里面不显示cpu温度了，cpu：j4125

veik, 2022-06-19:
我这里同时也不显示interfaces，N5105

656567543, 2022-06-19:
> 我这里同时也不显示interfaces，N5105

好像是浏览器的问题，你换个浏览器试下

veik, 2022-06-19:
> > 我这里同时也不显示interfaces，N5105
> 
> 好像是浏览器的问题，你换个浏览器试下

Edge, Chrome, Firefox都换了，我这里都一样

veik, 2022-06-20:
> > 我这里同时也不显示interfaces，N5105
> 
> 好像是浏览器的问题，你换个浏览器试下

更新了今天的版本，在Edge上可以看到了

656567543, 2022-06-21:
> > > 我这里也显示interfaces，N5105
> > 
> > 
> > 是浏览器的问题，你换个浏览器试下
> 
> 更新了今天的版本，在Edge上可以看到了

我的把那个插件更新下，又能显示了

DearTanker, 2022-06-21:
@656567543 插件名叫啥

DearTanker, 2022-06-21:
@kiddin9 隐身模式还是没有。F12 里面禁用缓存打勾以后刷新也还是没有。

kiddin9, 2022-06-21:
autocore 更新到最新版  命令行执行 cpuinfo 看看显示什么.

DearTanker, 2022-06-21:
@kiddin9 发现 `autocore-arm` 不知道为啥默认没安装，安装以后 CPU 占用率可以实时变化了，频率和温度的信息项也有了。
甚至 https://github.com/kiddin9/OpenWrt_x86-r2s-r4s-r5s-N1/issues/767 这个问题也解决了。。。

```
root@R4S 03:02 AM ~# cpuinfo
ARMv8 Processor x 6 (2016MHz 1608MHz, 45.6°C)
```

DearTanker, 2022-06-22:
@kiddin9 刚才升级了一下 Luci-Base 以后，概览里面又不显示 CPU 型号和端口速率了。。

DearTanker, 2022-07-26:
R4S 问题复现了。

概览不显示 CPU 温度，终端里面正常。

ASC8384, 2022-08-17:
> 我这里同时也不显示interfaces，N5105

您现在解决了吗