kiddin9/Kwrt #765: R5S固件网卡、语言问题

scegg, 2022-06-20:
使用0620编译直接eflasher固件开机刷入emmc的版本。

网卡工作在半双工模式：
[image]

客户端语言英语：主界面与多个界面内有非英语（也非中文）内容：
[image]
[image]

kiddin9, 2022-06-20:
语言问题后台 ctrl+f5 强制刷新下浏览器,
网卡是显示错误, 已经修复了, 软件包中安装ethtool 或更新最新固件即可.

scegg, 2022-06-20:
> 语言问题后台 ctrl+f5 强制刷新下浏览器, 网卡是显示错误, 已经修复了, 软件包中安装ethtool 或更新最新固件即可.

多谢。

我是5小时之前测试的（GMT+8 12:00PM）。如果是最近5小时内修改的，我明天再测一次。
再次感谢您的付出。

scegg, 2022-06-21:
Update:
今天新版固件，网卡问题已经解决。
语言问题依旧，CTRL+F5没有办法修复，语言设置为英语或者自动均会如此。