kiddin9/Kwrt #777: AX6版固件版本使用nginx编译后无法登陆后台管理界面

digiw, 2022-06-23:
AX6版固件版本使用nginx编译后无法登陆后台管理界面，求助！

kiddin9, 2022-06-23:
不能打开后台吗, 报什么错,

命令行 nginx -t -c /etc/nginx/uci.conf  看看,

digiw, 2022-06-23:
root@OpenWrt 01:28 PM ~# nginx -t -c /etc/nginx/uci.conf
Error loading shared library libubus.so.20220228: No such file or directory (needed by /usr/sbin/nginx)
Error loading shared library libubox.so.20211120: No such file or directory (needed by /usr/sbin/nginx)
Error loading shared library libblobmsg_json.so.20211120: No such file or directory (needed by /usr/sbin/nginx)
Error relocating /usr/sbin/nginx: blob_buf_init: symbol not found
Error relocating /usr/sbin/nginx: blobmsg_add_object: symbol not found
Error relocating /usr/sbin/nginx: blobmsg_open_nested: symbol not found
Error relocating /usr/sbin/nginx: blob_memdup: symbol not found
Error relocating /usr/sbin/nginx: blobmsg_add_json_element: symbol not found
Error relocating /usr/sbin/nginx: ubus_free: symbol not found
Error relocating /usr/sbin/nginx: blobmsg_parse_array: symbol not found
Error relocating /usr/sbin/nginx: ubus_lookup_id: symbol not found
Error relocating /usr/sbin/nginx: blob_nest_end: symbol not found
Error relocating /usr/sbin/nginx: blobmsg_add_field: symbol not found
Error relocating /usr/sbin/nginx: ubus_invoke_fd: symbol not found
Error relocating /usr/sbin/nginx: blob_buf_free: symbol not found
Error relocating /usr/sbin/nginx: blobmsg_parse: symbol not found
Error relocating /usr/sbin/nginx: blobmsg_format_json_with_cb: symbol not found
Error relocating /usr/sbin/nginx: ubus_connect: symbol not found
Error relocating /usr/sbin/nginx: ubus_lookup: symbol not found

kiddin9, 2022-06-23:
修复了,重新定制.

digiw, 2022-06-23:
感谢！只是我定制的时候它瞬间就显示编译完成，似乎没有再次编译。我有什么办法刷新它强制再编译一回？

kiddin9, 2022-06-23:
随便修改一点配置

digiw, 2022-06-23:
随意修改了一下初始登陆密码，可以再次编译！下载刷入后还是不能访问后台，再次用您之前的指令反馈结果如下：
root@OpenWrt 12:23 PM ~# nginx -t -c /etc/nginx/uci.conf
Error loading shared library libubus.so.20220228: No such file or directory (needed by /usr/sbin/nginx)
Error loading shared library libubox.so.20211120: No such file or directory (needed by /usr/sbin/nginx)
Error loading shared library libblobmsg_json.so.20211120: No such file or directory (needed by /usr/sbin/nginx)
Error relocating /usr/sbin/nginx: blob_buf_init: symbol not found
Error relocating /usr/sbin/nginx: blobmsg_add_object: symbol not found
Error relocating /usr/sbin/nginx: blobmsg_open_nested: symbol not found
Error relocating /usr/sbin/nginx: blob_memdup: symbol not found
Error relocating /usr/sbin/nginx: blobmsg_add_json_element: symbol not found
Error relocating /usr/sbin/nginx: ubus_free: symbol not found
Error relocating /usr/sbin/nginx: blobmsg_parse_array: symbol not found
Error relocating /usr/sbin/nginx: ubus_lookup_id: symbol not found
Error relocating /usr/sbin/nginx: blob_nest_end: symbol not found
Error relocating /usr/sbin/nginx: blobmsg_add_field: symbol not found
Error relocating /usr/sbin/nginx: ubus_invoke_fd: symbol not found
Error relocating /usr/sbin/nginx: blob_buf_free: symbol not found
Error relocating /usr/sbin/nginx: blobmsg_parse: symbol not found
Error relocating /usr/sbin/nginx: blobmsg_format_json_with_cb: symbol not found
Error relocating /usr/sbin/nginx: ubus_connect: symbol not found
Error relocating /usr/sbin/nginx: ubus_lookup: symbol not found
root@OpenWrt 12:24 PM ~#

closed 2022-06-24 by digiw