kiddin9/Kwrt #781: openssh 报错 fatal: privsep_preauth: preauth child terminated by signal 31

Editst, 2022-06-24:
https://github.com/coolsnowwolf/lede/issues/8262
错误和这个issue一致，期待能修复一下

Editst, 2022-06-24:
把libopenssh替换为`1.1.1n`可以暂时解决这个问题，期待能在编译时修复一下

Editst, 2022-07-09:
最新的`1.1.1p1`版本可以正常使用

closed 2022-07-09 by Editst

zanjie1999, 2022-08-15:
目前lede源上的软件包最新的openssh是9.0p1-1，openssl那三个包是1.1.1q-6，实测会出现这个问题

CHxCOOH, 2022-09-10:
非常感谢楼上！！这个问题困扰了我四五个小时，甚至换了固件…

我在新版固件里又遇到了同样问题。解决方案：安装和openssh版本匹配的openssl：`opkg install libopenssl-conf libopenssl1.1 openssl-util`。注意：软件源中的openssl版本可能高于openssh版本，请自行解决（手动升级openssh（P.s.比这新的正式版还没发）或者手动降级openssl）。经过测试，openssh版本8.9/9.0p1-1可以对应openssl1.1.1q-1，而更高版本的openssl1.1.1q-6是不可用的。

具体做法：先用sftp/web文件管理上传一份1.1.1q-1的ipk，可以在[清华镜像站](http://mirrors.tuna.tsinghua.edu.cn/openwrt/releases/22.03.0/packages/aarch64_generic/base/)下载，需要上述三个包，存下来传到路由器上。然后`rm /var/opkg-lists/*`删除版本信息，否则包管理器会自动下载更新的版本。然后卸载高版本：
```sh
opkg --force-depends remove libopenssl1.1
opkg --force-depends remove libopenssl-conf
opkg --force-depends remove openssl-util
```
（如果熟练的话也可以尝试直接覆盖安装，使用--force-downgrade）
再`opkg install`之前上传好的三个包就行了。