kiddin9/Kwrt #787: frps luci界面貌似不能用了，红米ax6

slslll, 2022-07-01:
主页下载的固件
[image]

然后在软件库中安装frps包如下
[image]

如图，frps界面中没有看到具体设置选项
[image]
[image]

kiddin9, 2022-07-01:
已修复. 更新软件列表后重装luci-app-frps

slslll, 2022-07-02:
好像没有变化，是我哪里没操作对吗

1
[image]

2
[image]

3
[image]

> 已修复. 更新软件列表后重装luci-app-frps

slslll, 2022-07-02:
难道是源的问题

slslll, 2022-07-02:
直接用主页定制化固件，这下可以了，感谢作者

closed 2022-07-02 by slslll

yu1chaofan, 2023-01-25:
[image]
[image]

同样显示不出来 
x86平台  固件是2022.11.28 以后的固件

zhibinlin1, 2023-02-12:
R2S遇到同样问题