kiddin9/Kwrt #794: AX6 更新至最新编译版本后，系统无法在线更新与云音乐解除无法启用

digiw, 2022-07-08:
1. 解除网易云音乐播放限制 ，插件始终无法启动
2. 服务：系统在线更新，提示：无法确认型号，或是不支持）之前是能提示版本是否是最新的。

digiw, 2022-07-08:
> 是今天定制编译的吗

对，今天编译的，另发现更新软件包会提示系统容量不足，会失败一批包的更新。
此外内存占用从更新完毕后的23%会持续上升，目前达到28%，正在继续观察中

kiddin9, 2022-07-08:
你确定固件版本日期是 7.08吗, 我刚刷了最新定制的, 在线更新功能没问题.
不扩容的话AX6只有30M左右的可用空间, 建议扩容或者刷uboot,有120M可用.

digiw, 2022-07-09:
经过仔细比对，发现这样几个问题，
1、我通过uboot 重刷过渡factory.ubi再更新到07.09最新版本，“系统在线更新”功能提示正常！

2、系统》软件包 页面，显示：已用空间4.1MB/6.7MB（61%），这个容量不是很理解系统只是什么，缓存？更新软件包时，是这个6.7MB容量满，所以提示包的更新失败，空间不足，但这里怎么清出空间呢，还请指点解惑，不然每次要重刷一次固件很蛋疼。

3、内存占用会缓慢上升，目前从开机的23%上升到25%，继续观察中

4、网易云音乐这个插件，发现两处问题：
     一是，在网页编译固件时，展开自定义列表，在出厂软件包列表内并没有luci-app-unblockneteasemusic 这个包，但下方 “所需软件包 (包名用空格隔开)” 这个文字列表框里却自带此插件；
     这个时候，在网页：常用软件包列表里，点选插件 “luci-app-unblockmusic 解锁网易云音乐” 后，下方 “所需软件包 (包名用空格隔开)” 文字列表里，就会多出一个：luci-app-unblockmusic 选中插件；
     完成编译后，openwrt的“服务”菜单里，就会出现两个云音乐解锁插件！

    二是，luci-app-unblockneteasemusic  这个云音乐解锁插件无法启用，在插件页面勾选启用保存，依旧显示未运行，此时如果到op网页后台的：系统：启动项里手动去启动或者重启这个插件，均提示失败；提示如下：
执行“/etc/init.d/unblockneteasemusic restart”失败：命令执行失败；
执行“/etc/init.d/unblockneteasemusic start”失败：命令执行失败

kiddin9, 2022-07-09:
[image]
你的设备有问题,应该是没扩容成功,我的还剩70多M的可用空间
框中的luci-app-unblockneteasemusic是你上次的记录,不需要的话删掉即可. 使用luci-app-unblockmusic就行.

digiw, 2022-07-09:
记得是扩容过的，我用指令看了下分区情况如下：
root@OpenWrt 02:41 PM ~# cat /proc/mtd
dev:    size   erasesize  name
mtd0: 00100000 00020000 "0:sbl1"
mtd1: 00100000 00020000 "0:mibib"
mtd2: 00300000 00020000 "0:qsee"
mtd3: 00080000 00020000 "0:devcfg"
mtd4: 00080000 00020000 "0:rpm"
mtd5: 00080000 00020000 "0:cdt"
mtd6: 00080000 00020000 "0:appsblenv"
mtd7: 00100000 00020000 "0:appsbl"
mtd8: 00080000 00020000 "0:art"
mtd9: 00080000 00020000 "bdata"
mtd10: 00080000 00020000 "crash"
mtd11: 00080000 00020000 "crash_syslog"
mtd12: 06640000 00020000 "rootfs"
mtd13: 00080000 00020000 "rsvd0"
------------------------------------------
求教，这分区是哪里姿势不太对？

digiw, 2022-07-09:
这样算起来 mtd12 rootfs 分区容量是102MB，很足够，挠头... 我这个问题是出来哪里，还得向你就救

digiw, 2022-07-09:
接着我去op后台“系统”页面，查看到 磁盘空间 和 临时空间 的情况，磁盘空间很小，临时空间很大
[image]

ssh登陆后查看磁盘使用情况如下：
[image]