kiddin9/Kwrt #800: 5.15.45 升级到 5.15.50版本BUG反馈

sniper3209, 2022-07-10:
目前是自定固件，采用线上固件升级时，原勾选的KMS插件会被bypass插件替换，5.15.50全锥型NAT无法正常开启，即使是在“网络-防火墙-常规设置”中选中“全锥型NAT”保存设置后，Turbo ACC 网络加速设置内仍显示“全锥型 NAT | 未运行”
后重新生成自定固件，重刷了一次，KMS恢复，全锥型NAT仍显示未运行。

NothingNA, 2022-07-11:
+1 全锥行nat一直无法运行~7.11重新编译后Turbo ACC里有全锥型 NAT选项了 勾选后依旧未运行……不知道7.10更新的两版turbo acc是不是都有啥问题

Pacio, 2022-07-11:
我的问题更多，在线编译的固件刷入以后直接lan口失联

sniper3209, 2022-07-11:
> +1 全锥行nat一直无法运行~7.11重新编译后Turbo ACC里有全锥型 NAT选项了 勾选后依旧未运行……不知道7.10更新的两版turbo acc是不是都有啥问题

其实是正常运行的，具体可以用NATtypetester进行测试，BUG只是在web界面显示未运行，不影响正常使用

sniper3209, 2022-07-11:
> 我的问题更多，在线编译的固件刷入以后直接lan口失联

再仔细些，有可能是LAN和WAN的位置互换了，5.15.50的固件我刷完后，路由IP地址的获取速度有点慢，等一会就可以正常登入了

Pacio, 2022-07-11:
> > 我的问题更多，在线编译的固件刷入以后直接lan口失联
> 
> 再仔细些，有可能是LAN和WAN的位置互换了，5.15.50的固件我刷完后，路由IP地址的获取速度有点慢，等一会就可以正常登入了

不是，经过我仔细测试，就是内核版本的问题，有的源码库都直接回退到5.15.48了

NothingNA, 2022-07-12:
> > > +1 全锥行nat一直无法运行~7.11重新编译后Turbo ACC里有全锥型 NAT选项了 勾选后依旧未运行......不知道7.10更新的两版turbo acc是不是都有啥问题
> > 
> > 
> > 其实是正常运行的，具体可以用NATtypetester进行测试，BUG只是在web界面显示未运行，不影响正常使用
> 
> 下载NATtypetester和另外一个NAT类型测试的软件，结果都是Symmetric NAT，刷回9号编译的固件，再测就是Full Cone NAT，所以我这肯定是有问题的。

又重新编译了下，这回第一次检测是PortRestrictedCone，然后变成fullcone，重开软件依旧，看来是可以正常全锥型 NAT，但是有bug，稳定性也好。

kiddin9, 2022-07-13:
已修复.