kiddin9/Kwrt #811: x86设备安装7.12版本 socat 及 miniupnpd 无效

gzxiexl, 2022-07-13:
系统日志内大量报错：daemon.warn miniupnpd[15821]: unknown rule type 3

socat 设置了状态显示✔，但是实际端口转发无效
miniupnpd 无效，一条都没有

gzxiexl, 2022-07-13:
[image]
[image]

kiddin9, 2022-07-13:
我的 miniupnpd  是正常的,
[image]

gzxiexl, 2022-07-13:
我刚新装的7.12 开始upnp也是正常的 可能是后面我安装某个插件跟它产生冲突了吧 暂时回到6.18了

XYZliang, 2022-07-16:
> 我刚新装的7.12 开始upnp也是正常的 可能是后面我安装某个插件跟它产生冲突了吧 暂时回到6.18了

我刚升级最新版的时候upnp里也是有内容的正常，后来搞了半天docker和adguardhome，搞完去upnp里看一眼就发现也是全空了（我这pcdn好几个，正常情况下upnp有四五十条），后来无奈重刷最新版…最后感觉还是upnp或者全锥nat有问题，弄回618版本了 #820