kiddin9/Kwrt #816: got error for the new version luci-app-bypass in X86

edwin19861218, 2022-07-15:
Seems some related components error or missing. Is there a free way to install the last correct version?

Plz check the detail output when re-installing.
 
Installing luci-app-bypass (git-22.190.50904-8006ace) to root...
Downloading https://op.supes.top/packages/x86_64/kiddin9/luci-app-bypass_git-22.190.50904-8006ace_all.ipk
Configuring luci-app-bypass.
cfg203db7
Too many arguments

And the error log is:

2022-07-15 16:35:14 SSR server: Shadowsocks Server1 Started!
2022-07-15 16:35:14 Socks5 Node: Can't find VMESS program, start failed!
2022-07-15 16:35:14 Main Node: Can't find VMESS program, start failed!

kiddin9, 2022-07-15:
Reedit node or update nodes subscription.

edwin19861218, 2022-07-15:
U r right, thanks

closed 2022-07-15 by edwin19861218