kiddin9/Kwrt #820: X86 最新定制固件 全锥Nat不生效

XYZliang, 2022-07-16:
#809 有提到7.10号的固件全锥问题。
为了升级我重新定制了固件（官网显示版本7.14），从6.18版定制固件不保留配置刷入x86路由并从0配置，刷入后nat从全锥nat1变为全锥nat4（upnp开启且生效，防火墙全锥nat打开）
尝试过重新定制、恢复旧版备份、防火墙强开fullcore全不行。
因为旧版固件我也没保存，只能刷回官网通用的618固件，nat恢复到正常的全锥nat1
至少尝试重新定制了五六次，重新刷机九十次….也在这其中遇到了另一个很蛋疼的问题： #819 
麻烦大佬抽空修复一下

kiddin9, 2022-07-16:
装个luci-app-turboacc 看 fullconenat 是否显示运行中

XYZliang, 2022-07-16:
> 装个luci-app-turboacc 看 fullconenat 是否显示运行中

已经试过了，是显示开启的，也有关闭再重启，依旧是没有生效。包括有个叫luci-app-corefullnat什么来着的插件，也是尝试安装运行了，但是nat还是老样子

JiuyeXD, 2022-07-16:
好像是 firewall4 的问题 以前的规则对新的防火墙无效 我正在测试 如果有最新进展我会继续跟进回复

NothingNA, 2022-07-17:
我是711那版nat4 714的刚刷好开测试软件 nat3 然后再点test显示nat1 运行一段时间再用软件测 连哪个网址都是unkown 不过实际游戏感觉没啥差别 所以也不知道到底有没有生效

NothingNA, 2022-07-17:
啊 刚才刷回618那版玩了会大乱斗 延迟是13、14ms 昨天用新固件玩是22ms左右 还出现过一次卡顿后67ms的 
不确定是单纯的服务器波动还是固件NAT真的有问题

XYZliang, 2022-07-17:
> 啊 刚才刷回618那版玩了会大乱斗 延迟是13、14ms 昨天用新固件玩是22ms左右 还出现过一次卡顿后67ms的
> 不确定是单纯的服务器波动还是固件NAT真的有问题

因为我有不少pcdn，对nat的变化很敏感，网心云后台明显看出来在升级新固件后nat下降为端口限制，降级后为原来的全锥，所以应该是有问题的

XYZliang, 2022-07-17:
> 我是711那版nat4 714的刚刷好开测试软件 nat3 然后再点test显示nat1 运行一段时间再用软件测 连哪个网址都是unkown 不过实际游戏感觉没啥差别 所以也不知道到底有没有生效

unknown可以试试换stun软件和服务器，我记得有两三款，有一款的stun软件有国内小米和QQ的检测服务器，就很快

XYZliang, 2022-07-17:
> 好像是 firewall4 的问题 以前的规则对新的防火墙无效 我正在测试 如果有最新进展我会继续跟进回复

感谢各位大佬

XYZliang, 2022-07-17:
> 装个luci-app-turboacc 看 fullconenat 是否显示运行中

大哥不知道能不能定制旧版固件呢，618版本是很正常的，当时定制的固件删了😭

NothingNA, 2022-07-17:
> > 我是711那版nat4 714的刚刷好开测试软件 nat3 然后再点test显示nat1 运行一段时间再用软件测 连哪个网址都是unkown 不过实际游戏感觉没啥差别 所以也不知道到底有没有生效
> 
> unknown可以试试换stun软件和服务器，我记得有两三款，有一款的stun软件有国内小米和QQ的检测服务器，就很快

用的是NatTypeTester，有小米和QQ的检测，结果就是我说的那样，下午又上游戏测了下，应该是服务器波动，这个对玩游戏应该是没啥影响，可能会影响到别的吧

NothingNA, 2022-07-17:
> > 装个luci-app-turboacc 看 fullconenat 是否显示运行中
> 
> 大哥不知道能不能定制旧版固件呢，618版本是很正常的，当时定制的固件删了😭

因为上面有人说是firewall4的问题，所以我在软件列表里卸载了firewall4，然后安装了firewall，下载用软件测是正常的了~你可以试一下

NothingNA, 2022-07-17:
> > 装个luci-app-turboacc 看 fullconenat 是否显示运行中
> 
> 大哥不知道能不能定制旧版固件呢，618版本是很正常的，当时定制的固件删了😭

因为上面有人说是firewall4的问题，所以我在软件列表里卸载了firewall4，然后安装了firewall，然后用软件测是正常的了~
你可以试一下编译的时候-firewall4 然后再添加firewall

kiddin9, 2022-07-17:
已默认换成了 firewall 可在定制时选择firewall4

XYZliang, 2022-07-17:
> 已默认换成了 firewall 可在定制时选择firewall4

感谢，我明天试试看

XYZliang, 2022-07-18:
> 已默认换成了 firewall 可在定制时选择firewall4

已尝试重新定制，刷入后发现防火墙无全锥nat开关，此时网络为nat4。只能下载安装turboacc后开启全锥nat，网络变为nat1正常。
但出现另一个问题openwrt后台upnp里没有内容，类似 #811 ，实测upnp实际上成功了。刚刷入的固件没安装其他可能冲突的插件，都是定制时加入的：
`luci-app-frpc luci-app-netdata luci-app-netspeedtest luci-app-ramfree luci-app-sqm luci-app-ttyd luci-app-watchcat luci-app-adguardhome luci-app-serverchan luci-app-vlmcsd luci-app-wrtbwmon luci-app-cifs-mount luci-app-hd-idle luci-app-homebox luci-app-jd-dailybonus luci-app-qbittorrent luci-app-transmission luci-app-unblockmusic luci-app-wolplus luci-app-aria2 luci-app-cloudreve luci-app-attendedsysupgrade luci-app-homeassistant luci-app-nlbwmon luci-app-alist luci-app-rclone luci-app-tencentddns my-autocore-x86 luci-app-dockerman`

XYZliang, 2022-07-18:
> > > 装个luci-app-turboacc 看 fullconenat 是否显示运行中
> > 
> > 
> > 大哥不知道能不能定制旧版固件呢，618版本是很正常的，当时定制的固件删了😭
> 
> 因为上面有人说是firewall4的问题，所以我在软件列表里卸载了firewall4，然后安装了firewall，然后用软件测是正常的了~ 你可以试一下编译的时候-firewall4 然后再添加firewall

确实是firewall4的问题。但是莫名其妙我upnp页面里没数据但实际又是成功了

Seas0, 2022-07-21:
可以参考[这里](https://github.com/fullcone-nat-nftables/openwrt-firewall4-with-fullcone)对firewall4进行Patch

XYZliang, 2022-07-23:
> 这里

哎，只能下旧版普通固件自己opkg install了

junke, 2022-07-28:
> > 这里
> 
> 哎，只能下旧版普通固件自己opkg install了

LZ最后解决了吗