kiddin9/Kwrt #852: 0802这个版本是不是有问题，不能分配ip

nonozone, 2022-08-02:
路由是一台J4125，之前用7.12号的正常使用，昨天看到更新，就急匆匆直接web升级，然后就无法上网了。折腾了一天，比如rufus直接刷新固件到U盘，用U盘启动也不行。用winpe启动之后，用写盘软件也不行，表现就是lan口获取不到ip地址。

实在没辙，又重新刷回7.12的版本，总算是可以了。

kiddin9, 2022-08-02:
什么网卡.

nonozone, 2022-08-02:
那个什么i255

kiddin9, 2022-08-02:
确实有问题,,  正在重新编译.

kiddin9, 2022-08-02:
已修复, 重新下载最新固件.

lukemin, 2022-08-02:
> 已修复, 重新下载最新固件.

redmi ax6  0802反复重启

kiddin9, 2022-08-02:
> > 已修复, 重新下载最新固件.
> 
> redmi ax6 0802反复重启

我的正常, 多等等,
[image]

Pacio, 2022-08-02:
0802不敢重启了，刚试了一下，重启以后等半天才能连上lan口，感觉还是有bug，不过比之前那个5.15.50好一些，起码等等能连上

AXBNG, 2022-08-02:
> > > 已修复, 重新下载最新固件.
> > 
> > 
> > redmi ax6 0802反复重启
> 
> 我的正常, 多等等, [image]

感觉还是有问题，恩山有人反馈
[image]

kiddin9, 2022-08-03:
正在修复, 目前已回退到上个版本,

sky540, 2022-08-05:
这个bug好像跟我上次反应的不能上网一样

Mioscout, 2022-08-12:
R2C似乎也有这个问题，目前已回退712正常使用