kiddin9/Kwrt #864: 今天云编译ax6，查看编译日志的错误内容

digiw, 2022-08-05:
反馈K大，不知道对固件本身使用有没有问题？

如下：
---------------

Calculating checksums...
错误日志:

Downloading file://www/wwwroot/op.supes.top/releases/targets/ipq807x/generic/5.15.45/Packages
Updated list of available packages in /www/wwwroot/op.supes.top/cache2/cache/ipq807x/generic/build_dir/target-aarch64_cortex-a53_musl/root-ipq807x/../../../../../../../../../../www/wwwroot/op.supes.top/cache2/cache/ipq807x/generic/dl/openwrt_core
Downloading file://www/wwwroot/op.supes.top/packages/aarch64_cortex-a53/base/Packages
Updated list of available packages in /www/wwwroot/op.supes.top/cache2/cache/ipq807x/generic/build_dir/target-aarch64_cortex-a53_musl/root-ipq807x/../../../../../../../../../../www/wwwroot/op.supes.top/cache2/cache/ipq807x/generic/dl/openwrt_base
Downloading file://www/wwwroot/op.supes.top/packages/aarch64_cortex-a53/packages/Packages
Updated list of available packages in /www/wwwroot/op.supes.top/cache2/cache/ipq807x/generic/build_dir/target-aarch64_cortex-a53_musl/root-ipq807x/../../../../../../../../../../www/wwwroot/op.supes.top/cache2/cache/ipq807x/generic/dl/openwrt_packages
Downloading file://www/wwwroot/op.supes.top/packages/aarch64_cortex-a53/luci/Packages
Updated list of available packages in /www/wwwroot/op.supes.top/cache2/cache/ipq807x/generic/build_dir/target-aarch64_cortex-a53_musl/root-ipq807x/../../../../../../../../../../www/wwwroot/op.supes.top/cache2/cache/ipq807x/generic/dl/openwrt_luci
Downloading file://www/wwwroot/op.supes.top/packages/aarch64_cortex-a53/routing/Packages
Updated list of available packages in /www/wwwroot/op.supes.top/cache2/cache/ipq807x/generic/build_dir/target-aarch64_cortex-a53_musl/root-ipq807x/../../../../../../../../../../www/wwwroot/op.supes.top/cache2/cache/ipq807x/generic/dl/openwrt_routing
Downloading file://www/wwwroot/op.supes.top/packages/aarch64_cortex-a53/Packages
Updated list of available packages in /www/wwwroot/op.supes.top/cache2/cache/ipq807x/generic/build_dir/target-aarch64_cortex-a53_musl/root-ipq807x/../../../../../../../../../../www/wwwroot/op.supes.top/cache2/cache/ipq807x/generic/dl/openwrt_kiddin9
Downloading file:packages/Packages
Updated list of available packages in /www/wwwroot/op.supes.top/cache2/cache/ipq807x/generic/build_dir/target-aarch64_cortex-a53_musl/root-ipq807x/../../../../../../../../../../www/wwwroot/op.supes.top/cache2/cache/ipq807x/generic/dl/imagebuilder
Collected errors:
* resolve_conffiles: Existing conffile /www/wwwroot/op.supes.top/cache2/cache/ipq807x/generic/build_dir/target-aarch64_cortex-a53_musl/root-ipq807x/etc/config/transmission is different from the conffile in the new package. The new conffile will be placed at /www/wwwroot/op.supes.top/cache2/cache/ipq807x/generic/build_dir/target-aarch64_cortex-a53_musl/root-ipq807x/etc/config/transmission-opkg.
Collected errors:
* opkg_conf_parse_file: Duplicate src declaration (openwrt_core https://op.supes.top/targets/ipq807x/generic/5.15.45). Skipping.
* opkg_conf_parse_file: Duplicate src declaration (openwrt_base https://op.supes.top/packages/aarch64_cortex-a53/base). Skipping.
* opkg_conf_parse_file: Duplicate src declaration (openwrt_packages https://op.supes.top/packages/aarch64_cortex-a53/packages). Skipping.
* opkg_conf_parse_file: Duplicate src declaration (openwrt_luci https://op.supes.top/packages/aarch64_cortex-a53/luci). Skipping.
* opkg_conf_parse_file: Duplicate src declaration (openwrt_routing https://op.supes.top/packages/aarch64_cortex-a53/routing). Skipping.
* opkg_conf_parse_file: Duplicate src declaration (openwrt_kiddin9 https://op.supes.top/packages/aarch64_cortex-a53/kiddin9). Skipping.
* check_data_file_clashes: Package luci-i18n-wizard-zh-cn wants to install file /www/wwwroot/op.supes.top/cache2/cache/ipq807x/generic/build_dir/target-aarch64_cortex-a53_musl/root-ipq807x/usr/lib/lua/luci/i18n/wizard.zh-cn.lmo
But that file is already provided by package * luci-app-wizard
* opkg_install_cmd: Cannot install package luci-i18n-wizard-zh-cn.
* check_data_file_clashes: Package luci-i18n-upnp-zh-cn wants to install file /www/wwwroot/op.supes.top/cache2/cache/ipq807x/generic/build_dir/target-aarch64_cortex-a53_musl/root-ipq807x/usr/lib/lua/luci/i18n/upnp.zh-cn.lmo
But that file is already provided by package * luci-app-upnp
* opkg_install_cmd: Cannot install package luci-i18n-upnp-zh-cn.
* check_data_file_clashes: Package luci-i18n-turboacc-zh-cn wants to install file /www/wwwroot/op.supes.top/cache2/cache/ipq807x/generic/build_dir/target-aarch64_cortex-a53_musl/root-ipq807x/usr/lib/lua/luci/i18n/turboacc.zh-cn.lmo
But that file is already provided by package * luci-app-turboacc
* opkg_install_cmd: Cannot install package luci-i18n-turboacc-zh-cn.
* check_data_file_clashes: Package luci-i18n-opkg-zh-cn wants to install file /www/wwwroot/op.supes.top/cache2/cache/ipq807x/generic/build_dir/target-aarch64_cortex-a53_musl/root-ipq807x/usr/lib/lua/luci/i18n/opkg.zh-cn.lmo
But that file is already provided by package * luci-app-opkg
* opkg_install_cmd: Cannot install package luci-i18n-opkg-zh-cn.
* check_data_file_clashes: Package luci-i18n-gpsysupgrade-zh-cn wants to install file /www/wwwroot/op.supes.top/cache2/cache/ipq807x/generic/build_dir/target-aarch64_cortex-a53_musl/root-ipq807x/usr/lib/lua/luci/i18n/gpsysupgrade.zh-cn.lmo
But that file is already provided by package * luci-app-gpsysupgrade
* opkg_install_cmd: Cannot install package luci-i18n-gpsysupgrade-zh-cn.
* check_data_file_clashes: Package luci-i18n-firewall-zh-cn wants to install file /www/wwwroot/op.supes.top/cache2/cache/ipq807x/generic/build_dir/target-aarch64_cortex-a53_musl/root-ipq807x/usr/lib/lua/luci/i18n/firewall.zh-cn.lmo
But that file is already provided by package * luci-app-firewall
* opkg_install_cmd: Cannot install package luci-i18n-firewall-zh-cn.
* check_data_file_clashes: Package luci-i18n-transmission-zh-cn wants to install file /www/wwwroot/op.supes.top/cache2/cache/ipq807x/generic/build_dir/target-aarch64_cortex-a53_musl/root-ipq807x/usr/lib/lua/luci/i18n/transmission.zh-cn.lmo
But that file is already provided by package * luci-app-transmission
* opkg_install_cmd: Cannot install package luci-i18n-transmission-zh-cn.
* check_data_file_clashes: Package luci-i18n-aria2-zh-cn wants to install file /www/wwwroot/op.supes.top/cache2/cache/ipq807x/generic/build_dir/target-aarch64_cortex-a53_musl/root-ipq807x/usr/lib/lua/luci/i18n/aria2.zh-cn.lmo
But that file is already provided by package * luci-app-aria2
* opkg_install_cmd: Cannot install package luci-i18n-aria2-zh-cn.
* check_data_file_clashes: Package luci-i18n-acme-zh-cn wants to install file /www/wwwroot/op.supes.top/cache2/cache/ipq807x/generic/build_dir/target-aarch64_cortex-a53_musl/root-ipq807x/usr/lib/lua/luci/i18n/acme.zh-cn.lmo
But that file is already provided by package * luci-app-acme
* opkg_install_cmd: Cannot install package luci-i18n-acme-zh-cn.
* check_data_file_clashes: Package luci-i18n-base-zh-cn wants to install file /www/wwwroot/op.supes.top/cache2/cache/ipq807x/generic/build_dir/target-aarch64_cortex-a53_musl/root-ipq807x/usr/lib/lua/luci/i18n/base.zh-cn.lmo
But that file is already provided by package * luci-base
* opkg_install_cmd: Cannot install package luci-i18n-base-zh-cn.
Collected errors:
* opkg_conf_parse_file: Duplicate src declaration (openwrt_core https://op.supes.top/targets/ipq807x/generic/5.15.45). Skipping.
* opkg_conf_parse_file: Duplicate src declaration (openwrt_base https://op.supes.top/packages/aarch64_cortex-a53/base). Skipping.
* opkg_conf_parse_file: Duplicate src declaration (openwrt_packages https://op.supes.top/packages/aarch64_cortex-a53/packages). Skipping.
* opkg_conf_parse_file: Duplicate src declaration (openwrt_luci https://op.supes.top/packages/aarch64_cortex-a53/luci). Skipping.
* opkg_conf_parse_file: Duplicate src declaration (openwrt_routing https://op.supes.top/packages/aarch64_cortex-a53/routing). Skipping.
* opkg_conf_parse_file: Duplicate src declaration (openwrt_kiddin9 https://op.supes.top/packages/aarch64_cortex-a53/kiddin9). Skipping.
/www/wwwroot/op.supes.top/cache2/cache/ipq807x/generic/build_dir/target-aarch64_cortex-a53_musl/root-ipq807x/etc/init.d/bootcount: line 22: syntax error near unexpected token `)'
/www/wwwroot/op.supes.top/cache2/cache/ipq807x/generic/build_dir/target-aarch64_cortex-a53_musl/root-ipq807x/etc/init.d/bootcount: line 22: ` qnap,301w)'
[... 24 lines cut ...]
[... cut ...]
Makefile:18: warning: ignoring old recipe for target '/www/wwwroot/op.supes.top/cache2/cache/ipq807x/generic/build_dir/target-aarch64_cortex-a53_musl/json_info_files/openwrt-08.05.2022-ipq807x-generic-qnap_301w-squashfs-sysupgrade.bin.json'
Pseudo file "/dev" exists in source filesystem "/www/wwwroot/op.supes.top/cache2/cache/ipq807x/generic/build_dir/target-aarch64_cortex-a53_musl/root-ipq807x/dev".
Ignoring, exclude it (-e/-ef) to override.
70878+1 records in
70879+0 records out
72580096 bytes (73 MB, 69 MiB) copied, 0.154926 s, 468 MB/s
/www/wwwroot/op.supes.top/cache2/cache/ipq807x/generic/staging_dir/host/bin/tar: Option --mtime: Treating date '@1655094027' as 2022-06-13 12:20:27

kiddin9, 2022-08-06:
没问题,不过ax6今天有更新,可以重新定制个.

digiw, 2022-08-06:
> 没问题,不过ax6今天有更新,可以重新定制个.

反馈新情况，ax6全新编译后，Turbo ACC里面没有全锥NAT开启选项，无法开启

closed 2022-08-06 by digiw