kiddin9/Kwrt #868: Ax6 8月7日编译更新分支: 22.03.0-rc6 内核: 5.15.49固件后的一些情况汇总

digiw, 2022-08-07:
》部分信息显示及功能异常 （固件编译时勾选了启用 firewall4 nftable）
1. 概览信息里，路由器端口不能显示实际的网口接插状态，且网口均为半双工，
   
[image]

2. 在smartdns设置页面，左上方无法显示插件的当前状态，始终为：正在收集数据....，且插件的模式设置成重定向53端口无效，只能设置成作为dnsmasq上游服务器；
 
[image]

3. 当修改并点击保存应用按钮时，偶有出现下图的错误提示：
[image]

4. 进入“状态 - 防火墙” 菜单时，界面会显示检测到旧版的规则，如下图：
[image]

5. turboacc 加速插件里能显示“全锥型NAT” 的项目状态，但没有控制勾选项，且始终是未运行状态，如果进入菜单“系统- 高级设置- 防火墙”，强行使用参数 option fullcone '1' 来开启，会发现依然不能开启NAT 1状态，turbo acc页面仍然显示未运行，使用第三方工具实测发现无法开启NAT 1，且重启路由器后，option fullcone 参数自动归零；
[image]

在 #820 的反馈里看到有聊到 https://github.com/kiddin9/OpenWrt_x86-r2s-r4s-r5s-N1/issues/820  不知道是否有关，还请K大团队看看上述情况，非常感谢！

kiddin9, 2022-08-07:
好的, 感谢反馈, 我测试下, 第二条smartdns是设置的问题, 第四条使用nftables就会有这个按钮提示的,用来打开iptables防火墙.

AXBNG, 2022-08-07:
另外sfe貌似无法生效，只有flow offloading......

digiw, 2022-08-08:
> 好的, 感谢反馈, 我测试下, 第二条smartdns是设置的问题, 第四条使用nftables就会有这个按钮提示的,用来打开iptables防火墙.

感谢k大秒回，smartdns的问题我跑去模块开发git反馈了情况 https://github.com/pymumu/smartdns/issues/1007

不知道它不能重定向，以及左上方运行状态无法侦测显示会不会也和firewall4有关，提供K大参考。

leganck, 2022-08-09:
我的ax6也是显示半双工，且测速也是半双工速度，2022-08-06 构建的固件

digiw, 2022-08-09:
> 我的ax6也是显示半双工，且测速也是半双工速度，2022-08-06 构建的固件

请问兄弟是怎么测试半双工状态的？我记得只有iperf2里-d 参数支持双向测试，iperf3不支持这个了。

leganck, 2022-08-09:
> > 我的ax6也是半双工，并且测速也是半双工的显示速度，2022-08-06
> 
> 记得请兄弟们是怎么双工状态的？记得只有iperf2里-d的参数支持我不测试半个测试，iperf3支持这个了。

用homebox的压力测试

digiw, 2022-08-09:
> > > 我的ax6也是半双工，并且测速也是半双工的显示速度，2022-08-06
> > 
> > 
> > 记得请兄弟们是怎么双工状态的？记得只有iperf2里-d的参数支持我不测试半个测试，iperf3支持这个了。
> 
> 用homebox的压力测试

兄弟，你是对的，确实是半双工状态，当我只跑半双工单边上传或下载时，千兆连接速度是达标的。

当homebox这个工具切换到高级配置后，同时start开始上传和下载测试时，那么上传/下载只有一侧能跑到相对较高速度，如下图：
[image]

看来半双工这个问题是石锤的！

kiddin9, 2022-08-09:
好的,我尽快修复下.

kiddin9, 2022-08-10:
已修复.

digiw, 2022-08-11:
> 已修复.

k大，似乎还有一些异常，我重新编辑了一楼，请参考！感谢

kiddin9, 2022-08-11:
好的,感谢反馈,我修复下.

kiddin9, 2022-08-11:
我刚测试了,
fullconenat默认就是开启的,
[image]
用的是firewall而非firewall4, firewall4没测试.

digiw, 2022-08-11:
> 我刚测试了, fullconenat默认就是开启的, [image] 用的是firewall而非firewall4, firewall4没测试.

那就对了，firewall4 的环境里，turbo acc 模块里面无法出现全锥型NAT的可选项，没切换到firewall4之前，这里的兼容性貌似一直没什么问题。

kiddin9, 2022-08-11:
firewall4 环境已修复. 重新下载最新固件.

digiw, 2022-08-18:
> 我的ax6也是显示半双工，且测速也是半双工速度，2022-08-06 构建的固件

兄弟，我今天编译的最新版本刷新后，测速结果更新在顶楼第一条里了，我发现按全双工的概念是指上下行数据同时在两个端口跑，这个好像没问题 ，不过我不管怎么测试，发现同时跑双边，上下行的速度相加才是千兆的带宽，这个情况你那边怎么样？

leganck, 2022-08-20:
> > 我的ax6也是显示半双工，且测速也是半双工速度，2022-08-06 构建的固件
> 
> 兄弟，我今天编译的最新版本刷新后，测速结果更新在顶楼第一条里了，我发现按全双工的概念是指上下行数据同时在两个端口跑，这个好像没问题 ，不过我不管怎么测试，发现同时跑双边，上下行的速度相加才是千兆的带宽，这个情况你那边怎么样？

最近没时间更新还没测试

leganck, 2022-08-21:
> > 我的ax6也是显示半双工，且测速也是半双工速度，2022-08-06 构建的固件
> 
> 兄弟，我今天编译的最新版本刷新后，测速结果更新在顶楼第一条里了，我发现按全双工的概念是指上下行数据同时在两个端口跑，这个好像没问题 ，不过我不管怎么测试，发现同时跑双边，上下行的速度相加才是千兆的带宽，这个情况你那边怎么样？

我这边网口带宽也还是这样

digiw, 2022-08-21:
> > > 我的ax6也是显示半双工，且测速也是半双工速度，2022-08-06 构建的固件
> > 
> > 
> > 兄弟，我今天编译的最新版本刷新后，测速结果更新在顶楼第一条里了，我发现按全双工的概念是指上下行数据同时在两个端口跑，这个好像没问题 ，不过我不管怎么测试，发现同时跑双边，上下行的速度相加才是千兆的带宽，这个情况你那边怎么样？
> 
> 我这测试了一下网口带宽也还是上下行加在一起是千兆

那咱们得到的结果一样，所以我不知道这是不是属于正常现象，如果全双工千兆同时跑满上下双边带宽就是一千兆，那就妥了，主要是没有其他兄弟关注过这个，恩山论坛似乎有过讨论，但也缺乏实测的结论，百度上也有说上下行跑满应该有2千兆带宽，因此我最近一直很迷惑！