kiddin9/Kwrt #879: 大佬，有点小问题请教一下

zcmi, 2022-08-13:
msg1500云编译后没有生成factory过度包，sysupgrade提示无法判断固件类型。E8820s有概率也不生成factory过度包。另外E8820s的这个源码好像对wifi的支持好一点，原来的openwrt源码会掉5G，不知道云编译的源码有没有合并。https://github.com/TsXor/my-openwrt-archive/tree/master。[来源应该是这个帖子](https://www.right.com.cn/forum/forum.php?mod=viewthread&tid=5203754&highlight=8820s)

firewall4对NAT1的支持好像不太行，感觉有时候可以，有时候不行，不过firewall正常。

kiddin9, 2022-08-13:
msg1500按照这个教程刷,https://www.right.com.cn/FORUM/forum.php?mod=viewthread&tid=5506650&page=1,里面有提供过渡文件
E8820s你那边测试过掉5G吗?
firewall4目前可能是有点兼容性问题,

zcmi, 2022-08-14:
> msg1500按照这个教程刷,[https://www.right.com.cn/FORUM/forum.php?mod=viewthread&tid=5506650&page=1,里面有提供过渡文件](https://www.right.com.cn/FORUM/forum.php?mod=viewthread&tid=5506650&page=1,%E9%87%8C%E9%9D%A2%E6%9 [...]

谢谢大佬，8820用了两天了没有掉5G，MSG1500刷入成功，不过还有KVR漫游好像只能开启K和R?目前K2p，newifi3，8820s，1500打开都是只能开启K，R好像也能开启，但v开启不了。这个是还没适配还是有BUG吗？

closed 2023-02-06 by zcmi