kiddin9/Kwrt #889: AX6S不定时自动重启

ERxhao, 2022-08-18:
`                                                 _____
      _____________________________      __________  /_
      _  __ \__  __ \  _ \_  __ \_ | /| / /_  ___/  __/
      / /_/ /_  /_/ /  __/  / / /_ |/ |/ /_  /   / /_
      \____/_  .___/\___//_/ /_/____/|__/ /_/    \__/
            /_/    
         _________________________________________
			   
               OpenWrt 08.02.2022 by Kiddin'
         _________________________________________
	 
系统负载:  0.33 0.25 0.21  	运行时间:  21 min		
内存已用:  21 % of 227MB  	IP  地址:  192.168.0.1
系统存储:  17% of 78.7M  	CPU 信息: ARMv8 Processor x 2	

[root@OpenWrt:02:28 PM ~] # cat /sys/fs/pstore/*
Oops#1 Part1
<6>[    9.048398] tun: Universal TUN/TAP device driver, 1.6
<6>[    9.074886] cryptodev: driver 1.12 loaded.
<6>[    9.081134] Loading modules backported from Linux version v5.15.58-0-g7d8048d4e064
<6>[    9.088728] Backport generated by backports.git v5.15.58-1-0-g42a95ce7
<6>[    9.116066] xt_time: kernel timezone is -0000
<7>[    9.172139] ieee80211 phy0: Selected rate control algorithm 'minstrel_ht'
<7>[    9.183164] mt7915e 0000:01:00.0: assign IRQ: got 142
<7>[    9.188314] pci 0000:00:00.0: enabling bus mastering
<6>[    9.193291] mt7915e 0000:01:00.0: enabling device (0000 -> 0002)
<7>[    9.199395] mt7915e 0000:01:00.0: enabling bus mastering
<6>[    9.261106] mt7622-wmac 18000000.wmac: HW/SW Version: 0x8a108a10, Build Time: 20190801210006a
<6>[    9.261106] 
<6>[    9.366162] mt7622-wmac 18000000.wmac: N9 Firmware Version: _reserved_, Build Time: 20220630094834
<6>[    9.408501] mt7915e 0000:01:00.0: HW/SW Version: 0x8a108a10, Build Time: 20211222184017a
<6>[    9.408501] 
<6>[    9.549747] mt7915e 0000:01:00.0: WM Firmware Version: ____000000, Build Time: 20211222184052
<6>[    9.614454] mt7915e 0000:01:00.0: WA Firmware Version: DEV_000000, Build Time: 20211222184111
<6>[    9.735732] PPP generic driver version 2.4.2
<6>[    9.740941] PPP MPPE Compression module registered
<6>[    9.746777] NET: Registered protocol family 24
<14>[    9.753146] kmodloader: done loading kernel modules from /etc/modules.d/*
<6>[   10.355906] zram0: detected capacity change from 0 to 78643200
<6>[   10.426265] Adding 76796k swap on /dev/zram0.  Priority:100 extents:1 across:76796k SSDsc
<6>[   12.700070] mtk_soc_eth 1b100000.ethernet eth0: Link is Down
<6>[   12.703505] mtk_soc_eth 1b100000.ethernet eth0: configuring for fixed/2500base-x link mode
<6>[   12.703677] mtk_soc_eth 1b100000.ethernet eth0: Link is Up - 2.5Gbps/Full - flow control rx/tx
<6>[   12.703695] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
<6>[   12.707013] mt7530 mdio-bus:00 lan1: configuring for phy/gmii link mode
<6>[   12.707758] 8021q: adding VLAN 0 to HW filter on device lan1
<6>[   12.712393] br-lan: port 1(lan1) entered blocking state
<6>[   12.712404] br-lan: port 1(lan1) entered disabled state
<6>[   12.716849] device lan1 entered promiscuous mode
<6>[   12.716857] device eth0 entered promiscuous mode
<6>[   12.726779] mt7530 mdio-bus:00 lan2: configuring for phy/gmii link mode
<6>[   12.727334] 8021q: adding VLAN 0 to HW filter on device lan2
<6>[   12.731970] br-lan: port 2(lan2) entered blocking state
<6>[   12.731981] br-lan: port 2(lan2) entered disabled state
<6>[   12.732684] device lan2 entered promiscuous mode
<6>[   12.741533] mt7530 mdio-bus:00 lan3: configuring for phy/gmii link mode
<6>[   12.743953] 8021q: adding VLAN 0 to HW filter on device lan3
<6>[   12.748836] mt7530 mdio-bus:00 lan3: Link is Up - 1Gbps/Full - flow control rx/tx
<6>[   12.752498] br-lan: port 3(lan3) entered blocking state
<6>[   12.752509] br-lan: port 3(lan3) entered disabled state
<6>[   12.753896] device lan3 entered promiscuous mode
<6>[   12.756669] br-lan: port 3(lan3) entered blocking state
<6>[   12.756681] br-lan: port 3(lan3) entered forwarding state
<6>[   12.756768] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
<6>[   12.769739] mt7530 mdio-bus:00 wan: configuring for phy/gmii link mode
<6>[   12.770260] 8021q: adding VLAN 0 to HW filter on device wan
<6>[   12.774823] mt7530 mdio-bus:00 wan: Link is Up - 1Gbps/Full - flow control off
<6>[   12.777093] IPv6: ADDRCONF(NETDEV_CHANGE): wan: link becomes ready
<6>[   13.159944] pppoe-wan: renamed from ppp0
<6>[   13.433189] mt7530 mdio-bus:00 lan2: Link is Up - 100Mbps/Full - flow control rx/tx
[... 169 lines cut ...]
<7>[10902.021311] x19: dead0000000000f0 x18: 0000000000000000 
<7>[10902.026615] x17: 0000000000000015 x16: ffffffc0108185d0 
<7>[10902.031919] x15: 000000000000003d x14: 0000000000000000 
<7>[10902.037224] x13: ffffffc008ab9398 x12: ffffffffffffffff 
<7>[10902.042528] x11: 0000000000000008 x10: ffffff8000005100 
<7>[10902.047832] x9 : ffffff8003afff7c x8 : 0000000000000000 
<7>[10902.053136] x7 : ffffffffffffffff x6 : 0089e70200000000 
<7>[10902.058441] x5 : 0000000000000002 x4 : 0000000000000000 
<7>[10902.063744] x3 : ffffff800018ad00 x2 : 0000000000000000 
<7>[10902.069049] x1 : dead000000000100 x0 : 0000000000000001 
<7>[10902.074354] Call trace:
<7>[10902.076795]  nf_flow_offload_ip_hook+0x7c4/0x138c [nf_flow_table]
<7>[10902.082880]  nf_flow_table_offload_setup+0x478/0x670 [nf_flow_table]
<7>[10902.089230]  process_one_work+0x1d4/0x370
<7>[10902.093231]  worker_thread+0x178/0x4e0
<7>[10902.096971]  kthread+0x120/0x124
<7>[10902.100192]  ret_from_fork+0x10/0x18
<0>[10902.103761] Code: d1004013 eb0002bf 54000580 52800014 (f9401263) 
<4>[10902.109846] ---[ end trace 35a8132a39490859 ]---
<0>[10902.119936] Kernel panic - not syncing: Oops: Fatal exception
<2>[10902.125675] SMP: stopping secondary CPUs
<0>[10902.129591] Kernel Offset: disabled
<0>[10902.133073] CPU features: 0x0000002,04002004
<0>[10902.137333] Memory Limit: none
`

digiw, 2022-08-19:
如果安装了Watchcat 模块（服务菜单里），可以尝试把 默认的 8.8.8.8那个ip改成你能ping通的国内地址，就不会导致误判重启了，或者用不到的话你可以关闭这个模块，希望对你有帮助

ERxhao, 2022-08-19:
> 如果安装了Watchcat模块（菜单地址里），可以试一下8.8.8.8那个ip成你ping通的国内模块，就不会导致判别重启了，或者这个用不到的话你可以关闭，希望对你有帮助

谢谢你提供的方法，我的 模块只安装了openchash upnp bypass 和自带的系统在线更新

SRGOD, 2022-08-27:
同样ax6s，模块只装了 系统在线更新、bypass 和 upnp，也是不定时重启，每天大概重启3、4次。

dreamflandre, 2022-08-30:
尝试下关闭网络-防火墙-路由/NAT 分载？我之前也是没装啥也反复重启，关了就稳定多了.

SRGOD, 2022-08-30:
> 

我的 网络-防火墙 页面里面没有路由/Nat 分载 这一项啊..