kiddin9/Kwrt #897: 请教一个 nginx 配置的问题

CliffHan, 2022-08-23:
看到这个固件默认 http 服务器是 nginx，想请教一个问题。

如果我想用禁用 443 端口服务（其实是想把这个端口留给 openvpn 用），应该怎么配置呢？

感谢您的工作。

CliffHan, 2022-08-23:
找到了这个：https://openwrt.org/docs/guide-user/services/webserver/nginx
是不是 `uci set nginx._ssl=disable` ？

kiddin9, 2022-08-23:
修改 /etc/config/nginx 文件中的443端口

CliffHan, 2022-08-23:
> 修改 /etc/config/nginx 文件中的443端口

我在高级设置中，看到了这一部分。
然后我改成了这样：

```
config main global
        option uci_enable 'true'

config server '_lan'
        list listen '80 default_server'
        list listen '[::]:80 default_server'
        option server_name 'op'
        list include 'conf.d/*.locations'
        option access_log 'off; # logd openwrt'

config server '_ssl'
        #list listen '443 ssl'
        #list listen '[::]:443 ssl'
        #option server_name '_ssl'
        #list include 'conf.d/*.locations'
        #option ssl_certificate '/etc/nginx/conf.d/_lan.crt'
        #option ssl_certificate_key '/etc/nginx/conf.d/_lan.key'
        #option ssl_session_cache 'shared:SSL:32k'
        #option ssl_session_timeout '64m'
        #option access_log 'off; # logd openwrt'

config server '_redirect2ssl'
        #list listen '80'
        #list listen '[::]:80'
        #option server_name '_redirect2ssl'
        #option return '302 https://$host$request_uri'
```

但重启后还是不对，443端口仍然被占用：netstat -anp 可以看到：
```
tcp        0      0 0.0.0.0:443             0.0.0.0:*               LISTEN      1470/uci.conf -g da
```

CliffHan, 2022-08-23:
找到了，/etc/nginx/conf.d/shortcuts.conf 中间有个 PVE 部分也在监听 443，挡掉那个就好了，谢谢。

closed 2022-08-23 by CliffHan