kiddin9/Kwrt #914: 感谢大佬的自定义固件，关于京东云RE-SP-01B系统存储空间问题

kerncoola, 2022-08-29:
[image]
[image]

系统安装后软件包安装空间只有10多兆，无限宝自带64G内置存储，能不能利用上，我也试过后期挂在到overlay中，但多次重启后进不去系统，想着系统安装时就分好区更稳吧。

谢谢。

kerncoola, 2022-08-29:
[image]

kiddin9, 2022-08-29:
这个内置存储只能通过挂载为其他分区使用,要作为overlay的话比较麻烦,你可以参考esir相关的视频.
https://www.youtube.com/watch?v=YwbwzuXKNlg

kerncoola, 2022-08-30:
OK

closed 2022-08-30 by kerncoola