kiddin9/Kwrt #929: 更新失败

716c, 2022-09-06:
Collected errors:
 * opkg_download: 下载失败 https://op.supes.top/targets/x86/64/5.15.63/Packages.gz, wget returned 5.
 * opkg_download: 下载失败 https://op.supes.top/packages/x86_64/base/Packages.gz, wget returned 5.
 * opkg_download: 下载失败 https://op.supes.top/packages/x86_64/packages/Packages.gz, wget returned 5.
 * opkg_download: 下载失败 https://op.supes.top/packages/x86_64/luci/Packages.gz, wget returned 5.
 * opkg_download: 下载失败 https://op.supes.top/packages/x86_64/routing/Packages.gz, wget returned 5.
 * opkg_download: 下载失败 https://op.supes.top/packages/x86_64/kiddin9/Packages.gz, wget returned 5.

kiddin9, 2022-09-06:
后台 网络诊断 测试下 op.supes.top 通不通

dolphin738, 2022-09-06:
是不是用AdGuard Home之类的过滤插件了?

716c, 2022-09-06:
> 是不是用AdGuard Home之类的过滤插件了?

是的DNS服务器是AdGuard

dolphin738, 2022-09-18:
在AdGuard设置 自定义过滤规则   @@||supes.top^

716c, 2022-09-18:
好的。谢谢

Sent from Mobile

Sent at: 2022年9月19日(Mon) Early Morning3:56
To: ***@***.***&gt;;
Cc: ***@***.******@***.***&gt;;