kiddin9/Kwrt #937: 红米AX6S 刷入底包后无限黄灯问题

MusherM, 2022-09-08:
https://supes.top/%e7%ba%a2%e7%b1%b3ax6s-%e8%a7%a3%e9%94%81ssh-%e5%88%b7openwrt%e6%95%99%e7%a8%8b/

在这篇教程下，我进行到了第四步，刷入OPENWRT正式版固件，根据https://supes.top/?version=22.03&target=mediatek%2Fmt7622&id=xiaomi_redmi-router-ax6s
“下载镜像”部分的指示，刷入了底包

于是就开始无限黄灯了，看评论区似乎很多人遇到了这个问题，github上issues里面好像又没人提到这个问题，请问该怎么解决呢？

kiddin9, 2022-09-08:
底包刷的是我教程中提供的吗?

kiddin9, 2022-09-08:
是刷入底包能打开后台,但是从底包中刷入正式固件无限黄灯吗

MusherM, 2022-09-08:
> 底包刷的是我教程中提供的吗?

所有文件均来源于
教程：https://supes.top/%e7%ba%a2%e7%b1%b3ax6s-%e8%a7%a3%e9%94%81ssh-%e5%88%b7openwrt%e6%95%99%e7%a8%8b/
固件下载：https://supes.top/?version=22.03&target=mediatek%2Fmt7622&id=xiaomi_redmi-router-ax6s

kiddin9, 2022-09-08:
图中的第二个固件就是底包呀,你要在底包中刷第一个固件,

MusherM, 2022-09-08:
只是教程第四步中的配图似乎直接刷入了
https://supes.top/?version=22.03&target=mediatek%2Fmt7622&id=xiaomi_redmi-router-ax6s
该页面的第一个固件，但根据下载页面提示，似乎应该先刷入第二个包

kiddin9, 2022-09-08:
过度包就是底包,只是不同叫法,

MusherM, 2022-09-08:
> 图中的第二个固件就是底包呀,你要在底包中刷第一个固件,

请问这个情况我能直接断电，使用小米官方的“小米路由器修复工具”进行修复吗？

kiddin9, 2022-09-08:
你直接断电重启试试能不能再进底包后台,不能的话就用小米路由器修复工具先刷回小米固件.

MusherM, 2022-09-09:
已经解决 感谢大大请问能否在教程或下载页强调一下 刷入过渡包（底包）后需要直接刷入正式版包 再刷一遍下载页的底包可能变砖发自我的 iPhone

lxp-git, 2023-07-23:
给还在搜这个问题的：试试官网的v23.05.0-rc1以上版本。具体问题的原因貌似是这两个大佬讨论的问题：https://github.com/openwrt/openwrt/pull/11814

我是一直用着官网的openwrt好好的，突然有一天重启了下开不了，然后拿官方修复工具修复了，但是不管刷什么openwrt的factory.bin都没办法进系统，刷完之后无限重启，蓝灯闪一下然后长亮黄灯，一直循环，像是在无限重启。