kiddin9/Kwrt #964: 发现一个很奇怪的问题，不知道是不是DNS的原因

zcmi, 2022-09-22:
科学上网的时候，访问不了谷歌和github，但youtube看8k都可以。google只能访问google.com.hk

同样的节点直接在电脑上使用或者同局域网中使用ImmortalWrt的newifi3和一个刷了你的固件的e8820s都可以正常访问google

目前msg1500主路由，N1和我家云作为旁路由使用时访问都有限制。但使用flippy固件的N1作为旁路由没有这些问题

查到v2ex上面有个人的nslookup也是差不多的情况，不过不清楚他用的什么固件。一开始以为是移动宽带的问题，后面换了newifi3，e8820s，flippy固件的N1可以访问google（使用你的固件时都基本是一模一样的设置，特地都只有安装bypass测试（因为其他几个软件都是一样））

这是google的测试和youtube的测试
[image]

[image]

[image]

下面是youtube的测试
[image]

[image]

borchian, 2022-09-23:
嗨，你好。我是HK1 box当旁路由，装了kiddin'的Opwrt系统。碰到一模一样的问题。一开始也是百思不得其解，在ssh openwrt ip下去nslookup www.google.com可以得到正常的回应，但是回到自己电脑端(windows, 或者 mac)去nslookup www.google.com得到的却是使导到内网的127.0.0.1。查到凶手原来是运营商的DNS污染手段造成。(运营商是中国移动，走他自己的DNS)
后来我把主路由(我的主路由是华硕的梅林固件)的DNS手动改成8.8.8.8，问题就解决了。

问题：我主路由的网关，内网DNS都已经导向旁路由。但就不知为何还是会用到主路由上设定的DNS。

目前只知道这是DNS污染造成。测试114.114.114.114，google.com一样会被导到127.0.0.1

zcmi, 2022-09-23:
谢谢兄弟，我待会试试，我是openwrt和Windows下nslookup都是指向127.0.0.1，昨天试过了114.114.114.114也是不行，以前没用这个固件时用过8.8.8.8，感觉有点延迟就没试。

zcmi, 2022-09-23:
> 嗨，你好。我是HK1 box当旁路由，装了kiddin'的Opwrt系统。碰到一模一样的问题。一开始也是百思不得其解，在ssh openwrt ip下去nslookup [www.google.com可以得到正常的回应，但是回到自己电脑端(windows](http://www.google.com%E5%8F%AF%E4%BB%A5%E5%BE%97%E5%88%B0%E6%AD%A3%E5%B8%B8%E7%9A%84%E5%9B%9E%E5%BA%94%EF%BC%8C%E4%BD%86%E6%98%AF%E5%9B%9E%E5%88%B0%E8%87%AA%E5%B7%B1%E7%94%B5%E8%84%91%E7%AB%AF(windows), 或者 mac)去nslookup [www.google.com得到的却是被导到内网的127.0.0.1。查到凶手原来是运营商的DNS污染手段造成。(运营商是中国移动，走他自己的DNS)](http://www.google.com%E5%BE%97%E5%88%B0%E7%9A%84%E5%8D%B4%E6%98%AF%E8%A2%AB%E5%AF%BC%E5%88%B0%E5%86%85%E7%BD%91%E7%9A%84127.0.0.1%E3%80%82%E6%9F%A5%E5%88%B0%E5%87%B6%E6%89%8B%E5%8E%9F%E6%9D%A5%E6%98%AF%E8%BF%90%E8%90%A5%E5%95%86%E7%9A%84DNS%E6%B1%A1%E6%9F%93%E6%89%8B%E6%AE%B5%E9%80%A0%E6%88%90%E3%80%82(%E8%BF%90%E8%90%A5%E5%95%86%E6%98%AF%E4%B8%AD%E5%9B%BD%E7%A7%BB%E5%8A%A8%EF%BC%8C%E8%B5%B0%E4%BB%96%E8%87%AA%E5%B7%B1%E7%9A%84DNS)) 后来我把主路由(我的主路由是华硕的梅林固件)的DNS手动改成8.8.8.8，问题就解决了。
> 
> 问题：我主路由的网关，内网DNS都已经导向旁路由。但就不知为何还是会用到主路由上设定的DNS。
> 
> 目前只知道这是DNS污染造成。测试114.114.114.114，[google.com](https://google.com)一样会被导到127.0.0.1

谢谢兄弟，我待会试试，我是openwrt和Windows下nslookup都是指向127.0.0.1，昨天试过了114.114.114.114也是不行，以前没用这个固件时用过8.8.8.8，感觉有点延迟就没试。

zcmi, 2022-09-24:
> 嗨，你好。我是HK1 box当旁路由，装了kiddin'的Opwrt系统。碰到一模一样的问题。一开始也是百思不得其解，在ssh openwrt ip下去nslookup [...]

试了一下，好像又不是DNS的问题，换了之后可以用google，但是bing加载变得极慢，今天再测试的时候google又上不了了，离大谱。只能是明年换运营商测试一下了

PieShe, 2022-09-25:
旁路由防火墙设置劫持53端口了吗？墙外解析dns必须走代理，才能不被劫持，在代理里面把dns的ip 加到代理列表，要么主路由用DNS over HTTPS来解析页可以，国内阿里 114 dnspod的公共代理都能正确解析被墙的域名，只是不走代理请求的时候会被污染，用DNS over HTTPS可以避免，走代理解析的话时候直接用8.8.8.8就好了

zcmi, 2022-09-25:
> 

劫持了，DNS也试过了，不只是旁路由这主路由也不行呀，全局都不行。换其他固件了

borchian, 2022-09-27:
> 试了一下，好像又不是DNS的问题，换了之后可以用google，但是bing加载变得极慢，今天再测试的时候google又上不了了，离大谱。只能是明年换运营商测试一下了

朋友，我是这样测试的…
旁路由ssh 进去，nslookup www.google.com 211.136.192.6 (这ip是深圳移动的dns)，测出来的画面如下：
[image]

再nslookup用别的dns这样就清楚明了哪个dns在搞google的鬼了。
[image]

另外，是的，我的旁路由是有使用smartdns，将国外及国内的分别指派dns， 再加上主路由的网关+dns都导向旁路由，基本上是没再碰到什么奇怪的事了。固件我也没有另外再增设防火墙，或是DNS转发，都没有。

digiw, 2022-10-11:
> 朋友，我是这样测试的… 旁路由ssh 进去，nslookup [www.google.com](http://www.google.com) 211.136.192.6 (这ip是深圳移动的dns)，测出来的画面如下： [image]
>
> 再nslookup用别的dns这样就清楚明了哪个dns在搞google的鬼了。 [image] [...]

smartdns大更了，解决了53端口劫持失败问题

closed 2023-02-06 by zcmi