kiddin9/Kwrt #982: N5105 i225 CPU满100%

XYZliang, 2022-10-04:
今天早上定制后刷入的n5105 i225物理机 8G内存（之前是用的爱快），家里500M上下传，挂了一个网心云跑了大概200M上传，只开启了upnp，frp，turboacc（仅开启bbr和全锥nat）、adguardhome、serverchan微信推送。白天一切正常，内存占用1G不到，CPU10%不到。到了晚上突然serverchan提示cpu负载高，进入后台发现CPU直接100%，负载超过了8，ssh进去后top无异常（图一）
[image]
netdata显示softirq软中断非常高（图二）
[image]
重启后正常。
proc/interrupts信息如下
[image]
百度找到个帖子，挺多说固件问题的，能否麻烦修复一下？ @kiddin9

kiddin9, 2022-10-06:
这个情况有复现么,

XYZliang, 2022-10-13:
> 这个情况有复现么,

会重复出现。无规律。出现三次后我暂时刷了爱快等修复。