kiddin9/Kwrt #986: upnp功能无效

flyfishes, 2022-10-08:
红米AX6，开启upnp：
启动 UPnP 与 NAT-PMP 服务
启用 UPnP 功能
启用 NAT-PMP 功能

但是在活动的 UPnP 重定向下面一条记录也没有。

[image]

h6fYe34KFSiH, 2022-10-08:
一样的情况，系统日志
Sat Oct  8 19:33:47 2022 daemon.err miniupnpd[10747]: refresh_nft_cache: mnl_cb_run returned -1
Sat Oct  8 19:33:47 2022 daemon.err miniupnpd[10747]: refresh_nft_cache: mnl_cb_run returned -1

tudoubocai, 2022-10-09:
看看是不是你调换过网卡的，之前我也遇到相同的问题。我用minipcie网卡+板载网卡，minipcie网卡性能要差些。我就把默认的wan和lan调换，用minipcie做wan。调换之后miniupnp似乎没有识别到网卡变了，于是失效。冲刷系统不动网卡，就好了。

lele712, 2022-10-09:
r5s一样

656567543, 2022-10-12:
X86 一样的

xinyue603, 2022-10-12:
> 看看是不是你调换过网卡的，之前我也遇到相同的问题。我用minipcie网卡+板载网卡，minipcie网卡性能要差些。我就把默认的wan和lan调换，用minipcie做wan。调换之后miniupnp似乎没有识别到网卡变了，于是失效。冲刷系统不动网卡，就好了。

2022.10.10，定制的版本同样问题，不调换网卡也没用。x86、r68s，都试过了，没有用。

kiddin9, 2022-10-13:
upnp已修复, 删除 miniupnpd-nftables ,安装 miniupnpd-iptables 即可. 或者重新定制.

flyfishes, 2022-10-14:
> upnp已修复, 删除 miniupnpd-nftables ,安装 miniupnpd-iptables 即可. 或者重新定制.

没用啊，默认根本没安装miniupnpd-nftables，安装了以下几个：

#opkg list-installed |grep upnp
luci-app-upnp - git-23.263.50281-79ba59a
miniupnpd - 3.0.20170421-19
miniupnpd-iptables - 3022-08-31-4

xinyue603, 2022-10-14:
> upnp已修复, 删除 miniupnpd-nftables ,安装 miniupnpd-iptables 即可. 或者重新定制.
我两个都装了，upnp显示出来了，还用删除掉miniupnpd-nftables吗？

miniwater, 2022-10-20:
红米ax6 10月20号最新固件也是一样问题，启用 IGDv1 模式默认打了勾，不过依然看不到记录