kiddin9/Kwrt #994: r68s UPNP 服务无效

gpsligit, 2022-10-10:
20221010定制
问题：如题
日志：
Mon Oct 10 23:49:14 2022 daemon.notice miniupnpd[6885]: HTTP listening on port 5000
Mon Oct 10 23:49:14 2022 daemon.notice miniupnpd[6885]: HTTP IPv6 address given to control points : [fdf8:35b9:5a14::1]
Mon Oct 10 23:49:14 2022 daemon.notice miniupnpd[6885]: Listening for NAT-PMP/PCP traffic on port 5351
Mon Oct 10 23:49:14 2022 daemon.err miniupnpd[6885]: refresh_nft_cache: mnl_cb_run returned -1
Mon Oct 10 23:49:14 2022 daemon.err miniupnpd[6885]: refresh_nft_cache: mnl_cb_run returned -1
Mon Oct 10 23:49:14 2022 daemon.err miniupnpd[6885]: refresh_nft_cache: mnl_cb_run returned -1
Mon Oct 10 23:49:17 2022 daemon.err miniupnpd[6885]: refresh_nft_cache: mnl_cb_run returned -1
Mon Oct 10 23:49:17 2022 daemon.err miniupnpd[6885]: refresh_nft_cache: mnl_cb_run returned -1
Mon Oct 10 23:49:17 2022 daemon.err miniupnpd[6885]: refresh_nft_cache: mnl_cb_run returned -1
Mon Oct 10 23:49:17 2022 daemon.err miniupnpd[6885]: refresh_nft_cache: mnl_cb_run returned -1
Mon Oct 10 23:49:17 2022 daemon.err miniupnpd[6885]: refresh_nft_cache: mnl_cb_run returned -1
Mon Oct 10 23:49:17 2022 daemon.err miniupnpd[6885]: refresh_nft_cache: mnl_cb_run returned -1
Mon Oct 10 23:49:22 2022 daemon.err miniupnpd[6885]: refresh_nft_cache: mnl_cb_run returned -1

gpsligit, 2022-10-10:
构建日志：
Building images for rockchip - FastRhino R68S
Packages: luci-app-argon-config luci-app-attendedsysupgrade luci-app-bypass luci-app-ddns luci-app-diskman luci-app-nginx-manager luci-ssl-nginx luci-theme-argon autocore-arm automount base-files bash block-mount busybox ca-bundle coremark curl dnsmasq-full dropbear ds-lite e2fsprogs fdisk firewall fstools htop kernel kmod-drm-rockchip kmod-gpio-button-hotplug kmod-gpu-lima kmod-ip6-tunnel kmod-ipt-nat kmod-ipt-nat6 kmod-lib-zstd kmod-r8125 kmod-tcp-bbr libc libgcc libustream-wolfssl logd lsblk luci-app-autoreboot luci-app-cpufreq luci-app-firewall luci-app-opkg luci-app-upnp luci-base luci-compat luci-lib-fs luci-lib-ipkg mkf2fs mtd nano netifd odhcp6c odhcpd-ipv6only openssh-sftp-server opkg partx-utils ppp ppp-mod-pppoe procd procd-seccomp resolveip uboot-envtools uci uclient-fetch urandom-seed urngd wget-ssl zram-swap

xinyue603, 2022-10-12:
同样的问题，我也是r68s的，花钱定制的这个固件。upnp用不了，怎么试也没用，准备卖了用回1900了，固件还是用官方的吧~

gpsligit, 2022-10-12:
是的，r68s性能还是差了点，用回X86平台了。

从 Windows 版邮件发送

发件人: xinyue603
发送时间: 2022年10月12日 13:57
收件人: kiddin9/OpenWrt_x86-r2s-r4s-r5s-N1
抄送: gpsligit; Author
主题: Re: [kiddin9/OpenWrt_x86-r2s-r4s-r5s-N1] r68s UPNP 服务无效 (Issue #994)

同样的问题，我也是r68s的，花钱定制的这个固件。upnp用不了，怎么试也没用，准备卖了用回1900了，固件还是用官方的吧~

kiddin9, 2022-10-13:
upnp已修复, 删除 miniupnpd-nftables ,安装 miniupnpd-iptables 即可.

gpsligit, 2022-10-14:
Great，非常感谢~~

从 Windows 版邮件发送

发件人: kiddin9
发送时间: 2022年10月14日 7:58
收件人: kiddin9/OpenWrt_x86-r2s-r4s-r5s-N1
抄送: gpsligit; Author
主题: Re: [kiddin9/OpenWrt_x86-r2s-r4s-r5s-N1] r68s UPNP 服务无效 (Issue #994)

upnp已修复, 删除 miniupnpd-nftables ,安装 miniupnpd-iptables 即可.