kiddin9/Kwrt #1006: nginx修改配置后不生效

AndyWu0107, 2022-10-15:
我的固件：8月23日在线定制的r5s，使用nginx服务

我的操作步骤：
1) 进入 /etc/nginx/conf.d，修改shortcuts.conf保存并退出
2) nginx -s reload -c '/etc/nginx/uci.conf'

现象：
1) 尝试访问及ping后发现shortcuts.conf中新增或修改的快捷访问均未生效，ping新增的说无法解析地址，ping修改的则会返回修改前的老ip地址
2) 使用nginx -T '/etc/nginx/uci.conf'，检查结果语法是ok的，可以正确打印出修改后的shortcuts.conf中的内容
3) 使用 /etc/init.d/nginx reload (restart) 也无法让shortcuts.conf中的修改生效
4) 使用 /etc/init.d/nginx stop 会报错说 open() "/etc/nginx/nginx.conf" failed (2: No such file or directory)

请问我是否遗漏了什么？谢谢！

AndyWu0107, 2022-10-22:
补充：
看了官网文档后，我理了一下nginx配置文件的自动生成的逻辑，又做了一些尝试：
把/etc/conf/nginx中的server_name op改为opp；
service nginx reload; service nginx restart

现象：
/etc/nginx/uci.conf中的op自动变为了opp；
但还是无法通过新的opp访问管理页面，原先的op依然生效。

kiddin9, 2022-10-22:
重启下路由器
清理下浏览器缓存或者换个全新的浏览器测试.

AndyWu0107, 2022-10-22:
感谢回复。还有一个点我比较困惑想请教一下：
我在局域网中ping op的时候，是能够解析出op的ip地址的，但我貌似没有找到这个“dns”的过程（比如web设置页面上的dns挟持）。请问这个地址解析是怎么实现的？或许这也能帮我继续查找原因。

AndyWu0107, 2022-10-22:
> 重启下路由器 清理下浏览器缓存或者换个全新的浏览器测试.

已尝试，重启路由器后问题依旧，在openwrt自己的终端中ping opp也返回bad address，ping op则还是生效的。
重启后uci.conf中server_name是opp。

kiddin9, 2022-10-22:
要修改op这个快捷方式的话,需要修改/etc/init.d/dnsmasq 中的
$lanaddr op" >/tmp/hosts/shortcuts
为
$lanaddr opp" >/tmp/hosts/shortcuts
然后 service dnsmasq restart

AndyWu0107, 2022-10-23:
感谢回复。先dns解析内网域名-再将请求指向op-最后op的nginx负责处理反代，这样逻辑就顺了。修改后也确实生效了。

我也读了一下/etc/init.d/dnsmasq，发现会利用正则做一步检查/etc/nginx/conf.d/shortcuts.conf中的set $ip语句，如果set $ip后面跟的地址不是op的lan地址，就会强行把它改成op的lan地址，这样会造成两个问题：
1）如果不通过set $ip变量的方式来修改目标服务的proxy_pass地址，而是直接把地址填到proxy_pass里面，则/etc/init.d/dnsmasq会找不到set $ip关键字，进而不认识这一个server，也就不会把这个server加进dnsmasq的配置文件中。
2）如果通过set $ip变量的方式修改目标服务的proxy_pass地址，但这个地址并不是op的lan地址，则dnsmasq每次启动的时候都会强行把它改成op的lan地址。这样做应该是为了让所有内网域名的访问都指向op，毕竟这样才能让op的nginx收到这个请求，进而做后续的代理转发。

这样分析下来，或许我最初就误解了shortcuts.conf的设计目标和适用范围：原先的shortcuts.conf或许只适用于反代和op lan共用同一个ip地址的那些服务，而我想要反代的服务位于局域网内的其它服务器上。

我的op lan地址是192.168.5.1/24，我想代理192.168.5.110:8096上的jellyfin服务，通过域名jellyfin/来访问。在现有的逻辑和结构下，我的做法是：

```
# Jellyfin
server {
		listen 80;
		listen [::]:80;
		server_name jellyfin;
		location /{
set $ip 192.168.5.1;  # 保证dnsmasq的启动脚本能识别到set $ip关键词从而认识这个server；保证jellyfin域名首先被发到op上，从而nginx能接到请求
			proxy_pass $scheme://192.168.5.119:8096;  # 保证nginx能把jellyfin访问请求转发到正确的ip上
		}
	}
```
这样做其实还是有点丑，不知道大佬有没有别的意见？

liouxy, 2022-12-02:
> 我也读了一下/etc/init.d/dnsmasq，发现会利用正则做一步检查/etc/nginx/conf.d/shortcuts.conf中的set $ip语句，如果set $ip后面跟的地址不是op的lan地址，就会强行把它改成op的lan地址，这样会造成两个问题： 1）如果不通过set [...]

我也是想把无线路由加进去看到这个issue， dnsmasq那段代码的意思大概是取读shortcuts那个文件，然后给加进生成的hosts里面，所以可以另外在这里写一段，读另外一个shortcuts，然后照类似的逻辑写hosts，但我不会写shell，算了算了hhh

closed 2022-12-02 by AndyWu0107