kiddin9/Kwrt #1007: N5S用SD卡，刷最新squashfs-sysupgrade无法启动

limonene0x, 2022-10-17:
N5S用SD卡，刷最新squashfs-sysupgrade无法启动，该怎么刷最新版本到SD卡中呢？为什么现在没有sd版本的了？

kiddin9, 2022-10-17:
https://supes.top/nanopi-r5s%e7%ba%bf%e5%88%b7openwrt%e8%87%b3emmc%e6%95%99%e7%a8%8b/
参考这个教程.

limonene0x, 2022-10-18:
手头没有公对公USB线，过两天试试。现在是不可以刷到SD卡里，从SD卡启动了吗？

kiddin9, 2022-10-18:
可以SD卡启动的.直接写入sysupgrade到SD卡中即可.

limonene0x, 2022-10-18:
可是用balenaEtcher将最新定制r5s-squashfs-sysupgrade.img.gz刷入SD卡之后无法启动，刷回之前的定制r5s_sd.img.gz可以正常启动

alan21j, 2022-10-19:
r4s有同樣情況，好像裏面缺少了network

limonene0x, 2022-10-20:
刷包之后分区状态是这样的
[image]

ryan5u3n, 2022-10-31:
同问，为什么不出之前的***_sd.img.gz固件了，最近的都是***-squashfs-sysupgrade.img.gz。
体积都差了几倍大小，之前的通过友善官方固件的emmc刷机助手直接刷入emmc，顺利启动，现在的不行了。

zealic, 2022-12-08:
这个版本的固件有问题 `09.05.2022-openwrt-rockchip-armv8-friendlyarm_nanopi-r5s_sd.img.gz`，刷好以后无法启动。

closed 2022-12-17 by limonene0x