kiddin9/Kwrt #1021: X86拨号无限重启

Q852456q, 2022-10-24:
10.18.2022更新后X86在线编译版选择PPPoE会无限重启，17号编写的倒是没这个问题。

kiddin9, 2022-10-24:
哪个型号的网卡,

Q852456q, 2022-10-24:
英特尔i211

Q852456q, 2022-10-24:
> 哪个型号的网卡,

英特尔i211

pzqpeter, 2022-10-24:
i225网卡和5105的CPU在pppoe拨号时也会卡住，有问题

kiddin9, 2022-10-24:
好的,我修复下.

kiddin9, 2022-10-24:
intel网卡我只有个I218-V, 测试拨号是正常的,
不定制试试默认固件
https://supes.top/?version=22.03&target=x86%2F64&id=generic

superwallx, 2022-10-25:
N5105

kiddin9, 2022-10-25:
> N5105

不定制,用默认固件也会吗?

superwallx, 2022-10-25:
我现在试一下

superwallx, 2022-10-25:
不定制，默认没问题，没有重启情况

kiddin9, 2022-10-25:
把你定制的固件下载链接发我测试下.

superwallx, 2022-10-25:
https://op.supes.top/store/f7700c890f1dddb3013a4e746f14065d/openwrt-10.25.2022-x86-64-generic-ext4-combined-efi.img.gz

superwallx, 2022-10-25:
这是刚才重新编译的 还没来得及试
我现在试一下

kiddin9, 2022-10-25:
定制的时候试试不勾选 
USB有线网卡: 
无线网卡
这两个选项.

kiddin9, 2022-10-25:
就是无线网卡驱动可能导致无限重启.或者试试拔掉无线网卡,还会不会导致重启.

superwallx, 2022-10-25:
不太好拆，安装在主板上的，我先试一下不选择usb网卡和无线网卡刷一次试一下

superwallx, 2022-10-25:
只去掉USB有线网卡目前好用
那问题就出在USB有线网卡上  已解决

kiddin9, 2022-10-25:
好的,感谢测试,我排查下是哪个USB有线网卡驱动的问题,

superwallx, 2022-10-25:
也感谢作者😄

kiddin9, 2022-10-25:
已修复.重新定制即可.

Q852456q, 2022-10-26:
> 已修复.重新定制即可.

我已经重新定制固件，只勾选了无线网卡，但是依旧存在拨号无限重启的问题，我稍后会上传问题固件

Q852456q, 2022-10-26:
> 已修复.重新定制即可.

链接：https://pan.baidu.com/s/17UaAuh2qm5BzX7WqmDILCA 
提取码：w5ue 
--来自百度网盘超级会员V4的分享

kiddin9, 2022-10-26:
ctrl+F5强制刷新定制页面后,再重新定制下.