kiddin9/Kwrt #1030: X86 docker无法拉去镜像

mbax2zf2, 2022-10-25:
想要用docker安装青龙面板，但在拉取镜像时报错：

> [root@OpenWrt:04:12 AM ~] # docker pull whyour/qinglong
> Using default tag: latest
> latest: Pulling from whyour/qinglong
> 213ec9aee27d: Downloading 
> 47858aee13bf: Download complete 
> cfe0d6c6d05c: Download complete 
> 2df617b3dcd9: Download complete 
> 38da16a6ebe8: Download complete 
> 0999315cb4ac: Download complete 
> 0ceac6badbb3: Download complete 
> latest: Pulling from whyour/qinglong
> 213ec9aee27d: Pull complete 
> 47858aee13bf: Extracting [==================================================>]  673.2kB/673.2kB
> cfe0d6c6d05c: Download complete 
> 2df617b3dcd9: Download complete 
> 38da16a6ebe8: Download complete 
> 0999315cb4ac: Download complete 
> 0ceac6badbb3: Download complete 
> failed to register layer: operation not supported

尝试拉取nginx镜像，同样报错：

> [root@OpenWrt:04:13 AM ~] # docker pull nginx
> Using default tag: latest
> latest: Pulling from library/nginx
> e9995326b091: Pull complete 
> 71689475aec2: Extracting [==================================================>]  25.41MB/25.41MB
> f88a23025338: Download complete 
> 0df440342e26: Download complete 
> eef26ceb3309: Download complete 
> 8e3ed6a9e43a: Download complete 
> latest: Pulling from library/nginx
> e9995326b091: Pull complete 
> 71689475aec2: Extracting [==================================================>]  25.41MB/25.41MB
> f88a23025338: Download complete 
> 0df440342e26: Download complete 
> eef26ceb3309: Download complete 
> 8e3ed6a9e43a: Download complete 
> failed to register layer: operation not supported

kiddin9, 2022-10-25:
先挂载新分区,并设置docker根目录为新分区

参考教程
https://www.evernote.com/shard/s642/sh/35a02421-5ad1-44d7-b8f7-f745fe7ac28a/592e15f4a3f1e9da8459cf11bc2ec87c

closed 2022-10-28 by mbax2zf2

wxd327, 2023-03-14:
#

> 先挂载新分区,并设置docker根目录为新分区
> 
> 参考教程 https://www.evernote.com/shard/s642/sh/35a02421-5ad1-44d7-b8f7-f745fe7ac28a/592e15f4a3f1e9da8459cf11bc2ec87c

这么弄完，可以用命令行操作 拉取镜像，安装好docker后，docker的应用也可以正常运行。但一旦重启，镜像和容器程序就都消失了。