kiddin9/Kwrt #1035: ax9000编译，部分基础包缺失

ercJuL, 2022-10-26:
```
Collected errors:
* opkg_install_cmd: Cannot install package kmod-ath10k-ct.
* opkg_install_cmd: Cannot install package kmod-ath11k-pci.
* opkg_install_cmd: Cannot install package kmod-usb-dwc3-qcom.
* opkg_install_cmd: Cannot install package libwolfsslcpu-crypto.
```
部分日志
```
Downloading file:packages/ipq-wifi-xiaomi_ax9000_1_aarch64_cortex-a53.ipk
Package kernel (5.15.72-1-59516b242c1eab7618054fbe2df35739) installed in root is up to date.
Unknown package 'kmod-ath10k-ct'.
Installing kmod-ath11k-ahb (5.15.72+6.1-rc0+2022-10-12-1) to root...
Downloading file:packages/kmod-ath11k-ahb_5.15.72%2b6.1-rc0%2b2022-10-12-1_aarch64_cortex-a53.ipk

Downloading file:packages/kmod-qrtr-smd_5.15.72-1_aarch64_cortex-a53.ipk
Unknown package 'kmod-ath11k-pci'.
Installing kmod-gpio-button-hotplug (5.15.72-3) to root...

Installing kmod-usb-dwc3 (5.15.72-1) to root...
Downloading file:packages/kmod-usb-dwc3_5.15.72-1_aarch64_cortex-a53.ipk
Unknown package 'kmod-usb-dwc3-qcom'.
Installing kmod-usb3 (5.15.72-1) to root...
Downloading file:packages/kmod-usb3_5.15.72-1_aarch64_cortex-a53.ipk

Package libgcc1 (11.2.0-4) installed in root is up to date.
Package libustream-wolfssl20201210 (2022-01-16-868fd881-2) installed in root is up to date.
Unknown package 'libwolfsslcpu-crypto'.
Installing logd (2021-08-03-205defb5-2) to root...
Downloading file:packages/logd_2021-08-03-205defb5-2_aarch64_cortex-a53.ipk
Installing luci-app-advanced (1.20-17) to root...

```

另外问下 是不是因为这个包`kmod-usb-dwc3-qcom`没有ax9000的固件才不能识别u盘

kiddin9, 2022-10-26:
确实漏了,等我修复下.

kiddin9, 2022-10-27:
已修复.

closed 2022-10-28 by ercJuL