kiddin9/Kwrt #1064: firewall4版本下的miniupnpd ntftable内网不能用报is not suitable for external IP

skill7899, 2022-11-12:
Sat Nov 12 08:32:40 2022 daemon.info vnstatd[1659]: Info: Interface "pppoe-wan" found from database, starting monitoring.
Sat Nov 12 08:32:40 2022 daemon.info vnstatd[1659]: Info: Monitoring (2): pppoe-wan (1000 Mbit) eth0 (10000 Mbit)
Sat Nov 12 08:32:40 2022 daemon.notice miniupnpd[6422]: private/reserved address 10.8.8.120 is not suitable for external IP
Sat Nov 12 08:32:40 2022 daemon.notice miniupnpd[6422]: private/reserved address 10.8.8.120 is not suitable for external IP
Sat Nov 12 08:32:43 2022 daemon.notice miniupnpd[6422]: private/reserved address 10.8.8.120 is not suitable for external IP
Sat Nov 12 08:32:43 2022 daemon.notice miniupnpd[6422]: private/reserved address 10.8.8.120 is not suitable for external IP
Sat Nov 12 08:32:45 2022 daemon.notice miniupnpd[6422]: private/reserved address 10.8.8.120 is not suitable for external IP
Sat Nov 12 08:32:45 2022 daemon.notice miniupnpd[6422]: private/reserved address 10.8.8.120 is not suitable for external IP

fanghh6, 2022-12-03:
同这个问题，已经弃了，白捐了

skill7899, 2022-12-04:
自己编译加补丁就可以了