kiddin9/Kwrt #1088: 我家云L1pro风扇自动调整

lxcau, 2022-12-02:
大佬能否增加我家云L1pro的风扇自动调整功能，我家云的风扇有些吵。

之前刷其他armbian有这个功能，是用一个脚本读取温度来动态调整pwm输出控制风扇，我拷贝过来无法运行，脚本内容如下：

#!/bin/bash
echo $$ > /var/run/fan.pid

echo 85 > /sys/class/gpio/export
echo "high" > /sys/class/gpio/gpio85/direction
echo 85 > /sys/class/gpio/unexport

echo 79 > /sys/class/gpio/export
echo high > /sys/class/gpio/gpio79/direction
echo 0 > /sys/class/pwm/pwmchip0/export
echo 10000 > /sys/class/pwm/pwmchip0/pwm0/period
echo normal > /sys/class/pwm/pwmchip0/pwm0/polarity
echo 1 > /sys/class/pwm/pwmchip0/pwm0/enable
 
while true
do
	temp=$(cat /sys/class/thermal/thermal_zone0/temp)
	if [ $temp -gt 75000 ]; then
		echo 8000 > /sys/class/pwm/pwmchip0/pwm0/duty_cycle;
	elif  [ $temp -gt 65000 ]; then
		echo 5000 > /sys/class/pwm/pwmchip0/pwm0/duty_cycle;
	elif  [ $temp -gt 55000 ]; then
		echo 3000 > /sys/class/pwm/pwmchip0/pwm0/duty_cycle;
	else
		echo 1500 > /sys/class/pwm/pwmchip0/pwm0/duty_cycle;
	fi
	sleep 10
done

lxcau, 2022-12-03:
原来有自动调速脚本，位置在/usr/bin/pwm-fan.pl，但执行报错，在系统日志里发现：
Thu Jan 21 16:50:35 2016 daemon.notice procd: /etc/rc.d/S95done: Can't locate strict.pm in @INC (you may need to install the strict module) (@INC contains: /usr/lib/perl5/5.28) at /usr/bin/pwm-fan.pl line 3.
Thu Jan 21 16:50:35 2016 daemon.notice procd: /etc/rc.d/S95done: BEGIN failed--compilation aborted at /usr/bin/pwm-fan.pl line 3.

zcmi, 2022-12-31:
加一，不知道怎么样移植到openwrt

lxcau, 2023-02-27:
已解决，检查后发现使用的flippy大佬的内核，在dtb里面已经将风扇写入hwmon模块进行自动调速了，可以反编译dtb到dts然后修改cooling-levels的四个状态数值，来调整风扇的四个状态的默认转速（0-255），如下：
fan0: pwm-fan {
		compatible = "pwm-fan";
		#cooling-cells = <2>;
		pwms = <&pwm 0 10000 0>;
		cooling-levels = <0 102 170 230>;
	};

closed 2023-02-27 by lxcau

zcmi, 2024-08-22:
> 已解决，检查后发现使用的flippy大佬的内核，在dtb里面已经将风扇写入hwmon模块进行自动调速了，可以反编译dtb到dts然后修改cooling-levels的四个状态数值，来调整风扇的四个状态的默认转速（0-255），如下： fan0: pwm-fan { compatible = "pwm-fan"; #cooling-cells = <2>; pwms = <&pwm 0 10000 0>; cooling-levels = <0 102 170 230>; };

你好，内核文件的位置是boot目录下吗，好像dtb文件和flippy与ophub大佬的是一样的，而且在晶晨宝盒换了flippy大佬的内核也不行