kiddin9/Kwrt #1100: 能否支持友善电子 NanoPi R5C？

luangong, 2022-12-09:
NanoPi R5C 刚发布，SoC 跟 R5S/R66S/R68S 一样是 RK3568，相比 R5S 少了一个千兆网口，不支持 NVMe SSD，内存和 eMMC 容量也不一样，但相比 R66S/R68S 多了 HDMI 口，方便断网时进控制台调试诊断。

参考链接：

- https://www.friendlyelec.com/index.php?route=product/product&product_id=290
- https://wiki.friendlyelec.com/wiki/index.php/NanoPi_R5C

kiddin9, 2022-12-09:
好的,

kiddin9, 2022-12-09:
已添加R5C支持
https://supes.top/?target=rockchip%2Farmv8&id=friendlyarm_nanopi-r5c

luangong, 2022-12-09:
神速啊，赞👍

closed 2022-12-11 by luangong

i11cn, 2022-12-14:
亲，亲测定制出来的ROM启动失败

kiddin9, 2022-12-14:
> 

有可能只是灯不亮,接上电脑试试.

bg7pge, 2022-12-14:
没有看到R5C的选项啊

i11cn, 2022-12-15:
LAN口接电脑，接上uart线，能进入console，检查网络，定制ROM时指定的ip地址都没设置成功，所以不能给电脑分配IP，FriendlyWRT就能正常工作，所以我觉得应该是某些配置项有问题
12月9日的那个ROM和我自己定制的ROM都是这样的现象

i11cn, 2022-12-15:
很抱歉能提供的信息不多，因为当时急着用，看了一眼之后就刷了FriendlyWRT用上了
我尽量过几天把它换下来，那时再记录一下错误信息之类的发出来

anschein, 2023-01-31:
> 已添加R5C支持 https://supes.top/?target=rockchip%2Farmv8&id=friendlyarm_nanopi-r5c

大神我定制了R5C的固件刷进去，目前发现的问题一个是指示灯不亮，只有电源灯工作，另外一个是M.2 WIFI无法工作，有需要的话我帮忙抓日志

bearcolin, 2023-02-19:
M.2 无线网卡系统设备没有列出，单独安装了rtl8822ce-firmware驱动包也不行，官方固件网络设备里是有wlan0设备的，这个固件里没有。不知道是不是需要手动设置？