kiddin9/Kwrt #1103: DockerMan界面和之前不同，不知道插件是否正常

qwazx, 2022-12-09:
这次编译的新固件，DockerMan的界面变成了这样，我就不知道该怎么用了。
[image]

而很久之前的固件，编译出来的DockerMan在里面是这样的
[image]

我不知道是哪里出错，或者说选错了插件？但是我重新安装插件也还是新的界面，我不知道如何使用。请问是插件更新，还是哪里出了问题？

kiddin9, 2022-12-09:
一个是概述页面, 一个是配置页面,

qwazx, 2022-12-10:
> 一个是概述页面, 一个是配置页面,

感谢回复。不过我现在的固件，右边的docker没有别的子页面了，只有这个配置页面，是需要挂载独立的存储设备才可以是吗？
[image]

pzqpeter, 2022-12-10:
我也是这样，猜测是dockerd没有启动成功

kiddin9, 2022-12-10:
什么设备.

qwazx, 2022-12-10:
> 什么设备.

硬酷R1(https://wiki.ikoolcore.cn/)，刚发售的没多久。不过我用之前可以的软路由测试了下编译的固件，也是同样的只有配置页面，没有其他子页面

pzqpeter, 2022-12-11:
x86设备，n5105

pzqpeter, 2022-12-11:
网上查到的资料是，这个症状是docker daemon没有启动导致的

pzqpeter, 2022-12-11:
将版本降级至[luci-base_git-23.342.44516-f01ab55_x86_64.ipk]，就正常了，但是新编异的软件无法降级啊

qwazx, 2022-12-11:
> 将版本降级至[luci-base_git-23.342.44516-f01ab55_x86_64.ipk]，就正常了，但是新编异的软件无法降级啊

难受……我之前编译的固件是可以的，但是删了……

kiddin9, 2022-12-11:
已修复, 重新定制,或者在软件包中在线更新下luci-base

kiddin9, 2022-12-11:
更新完 ctrl+F5强制刷新下页面.

qwazx, 2022-12-11:
> 已修复, 重新定制,或者在软件包中在线更新下luci-base

辛苦大佬修复！十分感谢！

pzqpeter, 2022-12-11:
谢谢大佬，已经合适了

88860, 2022-12-22:
大佬，我也是这个问题，原版openwrt源码删掉luci-base用你库的luci-base编译会报错

88860, 2022-12-22:
大佬，我也是这个dockerman显示错误问题，原版openwrt源码删掉自带luci-base用你库的luci-base编译会报错，该如何操作呢