kiddin9/Kwrt #1121: 自助构建镜像，所有docker容器无法访问外网

lilith-rong, 2022-12-18:
下面是我写入的所需软件包
luci-app-syncdial luci-app-serverchan luci-app-qbittorrent luci-app-aria2 luci-app-adguardhome luci-app-acme luci-app-dockerman luci-app-wolplus ddns-scripts-cloudflare luci-app-ttyd luci-app-alist luci-app-arpbind cloudflared luci-app-zerotier rsync luci-app-easymesh luci-app-mwan3 luci-app-rclone wakeonlan tailscale luci-app-watchcat luci-app-ddns bind-host luci-app-attendedsysupgrade luci-app-wireguard luci-app-adbyby-plus -luci-app-gpsysupgrade luci-app-nlbwmon
其他选项的截图
[image]
构建出来的镜像，在docker容器之中无法访问外部ip，只能ping通本地ip

kiddin9, 2022-12-18:
不要选择firewall4 试试.

lilith-rong, 2022-12-18:
谢谢！果然是这个的问题，困扰我好几天了！
谢谢大佬的回答

closed 2022-12-18 by lilith-rong