kiddin9/Kwrt #1143: bypass无法使用，xray core下载失败

Purewhiter, 2022-12-31:
- bypass日志如下
2022-12-31 22:49:33 xray core download failed.
2022-12-31 22:47:53 Check IP/GFW files...
2022-12-31 22:47:53 Check network status...
2022-12-31 22:47:53 GFW List 已更新
2022-12-31 22:47:50 China IPv6 List 已更新
2022-12-31 22:47:49 China IPv4 List 已更新
- 前些天是可以使用的

lon91ong, 2023-01-05:
+1
有没有手动更新的方法先顶一顶

lon91ong, 2023-01-05:
在腾讯镜像服务器中找到一个[xray core的安装包](https://mirrors.cloud.tencent.com/openwrt/releases/packages-22.03/mipsel_24kc/packages/xray-core_1.7.0-1_mipsel_24kc.ipk), ipk文件
可以试试下面的命令：
```
cd /tmp
wget https://mirrors.cloud.tencent.com/openwrt/relea
ses/packages-22.03/mipsel_24kc/packages/xray-core_1.7.0-1_mipsel_24kc.ipk
tar zxvf xray-core_1.7.0-1_mipsel_24kc.ipk ./data.tar.gz && rm -rf xray-core_1.7.0-1_mipsel_24kc.ipk
tar -zxvf data.tar.gz ./usr/bin/xray -C /tmp/bypass/bin && rm -rf ./data.tar.gz
```

lon91ong, 2023-01-05:
应该是终极自己可控的解决办法了

从[yichya/openwrt-xray](https://github.com/yichya/openwrt-xray/releases)项目下载mipsel_24kc架构的预编译ipk文件
我选的1.64版本，1.65+提及增大明显就放弃了

用上面一条回复的方法解压，上传路由器赋权测试没有问题，在路由器用`tar -zcvf myray.tar.gz xray`压缩得到xray的压缩包

然后放到国内的代码托管平台上
修改`/usr/share/bypass/downbin`文件中的下载链接为自己的myray.tar.gz地址, 完工!

[蓝奏云myray](https://sobweb.lanzoue.com/ieRJP0k5vl0d) 密码:fghb

linwk24, 2023-01-05:
这问题16M闪存的路由器 都有这个问题应该插件升级导致体积增大 闪存不够,核心文件下载失败,无法运行插件,因为运行SS协议是可以的,运行VLESS协议却运行不了

linwk24, 2023-01-05:
参考以上方法：
1，下载 /usr/share/bypass下的 downbin显示的核心文件地址“xray.tar.gz”
while ! curl -m 18 -Lfso $K/xray.tar.gz http://static.supes.top/mipsel_24kc/xray.tar.gz;do
2，将下载的“xray.tar.gz”文件用winscp登录 上传到/tmp/bypass/bin 目录吧原来的文件删除或者覆盖
3，进入路由器bypass 选中个节点运行,然后重启路由器。
4，进入bypass 日志可看到成功加载运行，即使重启无数遍也不影响正常运行
2023-01-05 17:48:32 ChinaDNS : Start ChinaDNS-NG successful!
2023-01-05 17:48:32 SmartDNS : Use https://dns.alidns.com/dns-query to start in DoH mode (Domestic DNS)
2023-01-05 17:48:32 SmartDNS : Use 8.8.8.8,8.8.4.4 to start in TCP mode (Foreign DNS)
2023-01-05 17:48:29 Main Node: Xray 1.6.4 Started!
2023-01-05 17:48:28 Start iptables rules successful!
2023-01-05 17:48:11 Download IP/GFW files...
2023-01-05 17:48:11 Check IP/GFW files...
2023-01-05 17:48:11 Check network status...
------------------------------------------------------------------------------------------------------------------------------------------------------------
2023-01-05 17:48:03 xray core download successful.
2023-01-05 17:43:04 Download IP/GFW files...
2023-01-05 17:43:04 Check IP/GFW files...
2023-01-05 17:43:01 Check network status...

MatcherAny, 2023-01-07:
参考以上的讨论 改写了下`/usr/share/bypass/downbin`文件, 
```bash
#!/bin/sh

log(){
	echo "$(date +'%Y-%m-%d %H:%M:%S') $*" >> /var/log/bypass.log
}

K=/tmp/bypass/bin
kill -9 $(ps -w | grep $K | grep -v grep | awk '{print$1}') 2>/dev/null
mkdir -p $K
c=0
	while ! curl -m 18 -Lfso $K/xray-core.ipk https://mirrors.cloud.tencent.com/openwrt/releases/packages-22.03/mipsel_24kc/packages/xray-core_1.7.0-1_mipsel_24kc.ipk;do
		[ $c -eq 4 ] && {
		log "xray core download failed."
		exit 1
		} || let c++
		sleep 2
	done
	log "xray core download successful."
	tar -zxf $K/xray-core.ipk ./data.tar.gz -C $K/ && rm -rf $K/xray-core.ipk
	tar -zxf $K/data.tar.gz ./usr/bin/xray -C $K/ && rm -rf $K/data.tar.gz
	mv $K/usr/bin/xray $K && rm -rf $K/usr/
	chmod +x $K/xray
	ln -s $K/xray /usr/bin/xray 2>/dev/null
	log "xray core bin file successful."

[ $1 = 0 ] && /etc/init.d/bypass start &
```

xray-core_1.7.0-1_mipsel_24kc.ipk的文件内容是这样的
```
xray-core.ipk
  │ control.tar.gz
  │ debian-binary
  └─data.tar.gz
    ├─etc
    ├─lib
    └─usr
       └─bin
           xray  <----------
```

sone-miyuki, 2023-01-14:
可能网址被干扰了，挂另一个路由做前置后可以成功下载xray

fujie-xiyou, 2023-02-26:
> 参考以上的讨论 改写了下`/usr/share/bypass/downbin`文件,
> 
> ```shell
> #!/bin/sh
> 
> log(){
> 	echo "$(date +'%Y-%m-%d %H:%M:%S') $*" >> /var/log/bypass.log
> }
> 
> K=/tmp/bypass/bin
> kill -9 $(ps -w | grep $K | grep -v grep | awk '{print$1}') 2>/dev/null
> mkdir -p $K
> c=0
> 	while ! curl -m 18 -Lfso $K/xray-core.ipk https://mirrors.cloud.tencent.com/openwrt/releases/packages-22.03/mipsel_24kc/packages/xray-core_1.7.0-1_mipsel_24kc.ipk;do
> 		[ $c -eq 4 ] && {
> 		log "xray core download failed."
> 		exit 1
> 		} || let c++
> 		sleep 2
> 	done
> 	log "xray core download successful."
> 	tar -zxf $K/xray-core.ipk ./data.tar.gz -C $K/ && rm -rf $K/xray-core.ipk
> 	tar -zxf $K/data.tar.gz ./usr/bin/xray -C $K/ && rm -rf $K/data.tar.gz
> 	mv $K/usr/bin/xray $K && rm -rf $K/usr/
> 	chmod +x $K/xray
> 	ln -s $K/xray /usr/bin/xray 2>/dev/null
> 	log "xray core bin file successful."
> 
> [ $1 = 0 ] && /etc/init.d/bypass start &
> ```
> 
> xray-core_1.7.0-1_mipsel_24kc.ipk的文件内容是这样的
> 
> ```
> xray-core.ipk
>   │ control.tar.gz
>   │ debian-binary
>   └─data.tar.gz
>     ├─etc
>     ├─lib
>     └─usr
>        └─bin
>            xray  <----------
> ```

感谢分享  不过腾讯这镜像有点狗  只保留最新版本了  目前这个版本已经404了  需要替换为最新版的地址

microyin715, 2023-03-09:
#!/bin/sh

cd /tmp
wget https://fastly.jsdelivr.net/gh/kiddin9/bin@main/mipsel_24k/xray.tar.gz
tar -zxf xray.tar.gz   && rm -rf xray.tar.gz
chmod +x /xray
mkdir bypass && cd /tmp/bypass && mkdir bin && cd /tmp
cp -r  xray /tmp/bypass/bin/xray && rm -rf xray
chmod +x /tmp/bypass/bin/xray

microyin715, 2023-03-11:
我发现修改/usr/share/bypass/downbin中下载时间可以解决，部分网络下载慢指定时间没下载下来导致出错
curl -m 18和curl -m 15，适当加长，我改成curl -m 36和curl -m 30就没问题了