kiddin9/Kwrt #1154: AX6的2023-1-9日新固件底包问题咨询

digiw, 2023-01-09:
新固件下载list里多出了两个过渡底包：
REDMI_AX6 DSQUASHFS DFACTORY.UBI
REDMI_AX6-INITRAMFS-FACTORY.UBI

这两个过渡包有什么功能上的不同？分别适合在什么场景使用呢？

kiddin9, 2023-01-14:
底包显示问题已修复
第一次你下载的是你定制的固件, 刷新后显示的是默认固件,再下载就是默认固件了.
另外新版本固件需要在uboot中刷入INITRAMFS-FACTORY.UBI底包

JohnnyDu0815, 2023-01-15:
> 

你好，我今天用最新的过渡包刷1.15的固件，无法识别网关10.0.0.1

digiw, 2023-01-15:
> > 
> 
> 你好，我今天用最新的过渡包刷1.15的固件，无法识别网关10.0.0.1

你的意思是uboot成功刷新过渡固件ubi，重启后成功进去过渡固件，然后过渡固件刷1.15再重启后10.0.0.1无法访问？

JohnnyDu0815, 2023-01-15:
> 

是的

digiw, 2023-01-15:
> > 
> 
> 是的

不知道你下载的是定制固件还是默认固件，如果是默认固件那得反馈K大了，如果是定制固件，可以检查一下是否定制的过程里，自定义了与默认不同的登陆IP

digiw, 2023-01-16:
> 底包显示问题已修复 第一次你下载的是你定制的固件, 刷新后显示的是默认固件,再下载就是默认固件了. 另外新版本固件需要在uboot中刷入INITRAMFS-FACTORY.UBI底包

感谢K大讲解！！今天重新定制了一下，在下载时发现如下情况，想请k大看看能否安全刷新？
1. 定制完成下载时，发现ubi底包的sha256哈希码与实际文件的始终不对应，截图如下：
[image]

2. K大你说的 INITRAMFS-FACTORY.UBI 底包已经没有了，只有上述下载的 SQUASHFS-FACTORY.UBI，最后那个initramfs-kernel.bin（首次刷机可能会用到）如果点击下载，是404状态，无法下载：
[image]

closed 2023-02-28 by digiw