kiddin9/Kwrt #1182: 竞斗云AdGuard Home核心更新失败

YYplus, 2023-01-20:
/usr/bin/curl
Local version: v0.102.0
v6., cloud version: v0.107.22.
Updating core...
start download
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed

  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
 27 6624k   27 1807k    0     0  1254k      0  0:00:05  0:00:01  0:00:04 1408k
 86 6624k   86 5711k    0     0  2430k      0  0:00:02  0:00:02 --:--:-- 2603k
100 6624k  100 6624k    0     0  2589k      0  0:00:02  0:00:02 --:--:-- 2757k
download success start copy
start copy
AdGuardHome service disabled
Command failed: ubus call service delete { "name": "AdGuardHome", "instance": "nobackup" } (Not found)
Succeeded in updating core.
Local version: v0.107.22, cloud version: v0.107.22.

HotMaker, 2023-01-26:
R68S也是一样的问题，重启路由后显示AdGuard Home未运行，已重定向

YYplus, 2023-01-26:
> R68S也是一样的问题，重启路由后显示AdGuard Home未运行，已重定向

手动更新吧，更新完运行正常