kiddin9/Kwrt #1199: 軟件更新後部份位置出現如下錯誤，該如何手動修復

alan21j, 2023-02-03:
Function statements require a function name
  in http://192.168.2.254/luci-static/resources/network.js?v=git-23.365.31656-c3ae5db:?
  at http://192.168.2.254/luci-static/resources/luci.js?v=git-24.033.44588-94eb2a9:179:29
  at async Promise.all (index 2)
  at async Promise.all (index 6)

部分位置功能不能正常顯示
謝謝

alan21j, 2023-02-03:
發現是最新的luci-base出錯，應該怎樣迴歸舊版本呢？

Miroxyz, 2023-02-03:
参考这个，到他们自建的仓库下载历史版本安装，例子里的是x86/64的平台
https://github.com/kiddin9/OpenWrt_x86-r2s-r4s-r5s-N1/issues/1200#issuecomment-1415186460

alan21j, 2023-02-03:
發現就算安裝回舊版本，上面一樓的錯誤仍在持續。無法解決

yijiangchengming, 2023-02-03:
> 問題已解決，必須先opkg remove luci-base新版本，然後“重新啓動”，使用ssh進入再安裝較舊版本luci-base。

山重水复疑无路，柳暗花明又一村。

kiddin9, 2023-02-03:
已修复 重新定制下 不好意思

WPF0414, 2023-02-03:
> 已修复 重新定制下 不好意思

晚上又可以折腾了，哈哈哈，不然晚上都没玩的了。