kiddin9/Kwrt #1200: Generic x86/64 今日构建完成 提示SyntaxError

orthwang, 2023-02-03:
openwrt-02.03.2023-x86-64-generic-squashfs-combined-efi.img 请修复

Kaffu-Chino, 2023-02-03:
+1 我也是提示错误
状态-》概览
```
SyntaxError
Function statements require a function name
  in http://xxx.xxx.xxx.xxx/luci-static/resources/network.js?v=git-23.365.31656-c3ae5db:?
  at http://xxx.xxx.xxx.xxx/luci-static/resources/luci.js?v=git-24.033.44588-94eb2a9:179:29
  at async Promise.all (index 4)
```
系统-》系统
```
SyntaxError
Function statements require a function name
  in http://xxx.xxx.xxx.xxx/luci-static/resources/network.js?v=git-23.365.31656-c3ae5db:?
  at http://xxx.xxx.xxx.xxx/luci-static/resources/luci.js?v=git-24.033.44588-94eb2a9:179:29
  at async Promise.all (index 2)
  at async Promise.all (index 6)
```
网络-》接口，路由，DHCP/DNS、网络诊断、防火墙
```
SyntaxError
Function statements require a function name
  in http://192.168.3.5/luci-static/resources/network.js?v=git-23.365.31656-c3ae5db:?
  at http://192.168.3.5/luci-static/resources/luci.js?v=git-24.033.44588-94eb2a9:179:29
```

1225475505, 2023-02-03:
我出现的问题，与他们相同！

Miroxyz, 2023-02-03:
我在旧版本升级 luci-base 这个包出现了这个问题，全新编译也有这个问题。

Miroxyz, 2023-02-03:
暂时降级到历史版本可以解决

```shell
# 下载历史版本
wget https://op.supes.top/packages/x86_64/old/luci-base_git-24.021.55995-d924631_x86_64.ipk
# 删除现在的中文包，否则历史版本装不上
opkg remove luci-i18n-base-zh-cn
# 安装历史版本
opkg install luci-base_git-24.021.55995-d924631_x86_64.ipk --force-downgrade
```

WPF0414, 2023-02-03:
同出现此问题！

yydjh, 2023-02-03:
r2s也是一样的

alw-AYs, 2023-02-03:
同出现此问题

orthwang, 2023-02-03:
问题已经被修复了

closed 2023-02-03 by orthwang

charley008, 2023-02-08:
貌似还是出现这个问题。2023-2-8