kiddin9/Kwrt #1213: 今日更新软件包后，无法进入后台

1225475505, 2023-02-08:
今日更新了软件包中心的三个插件源，重启了还是无法进入后台，一直提示：正在载入视图…

griphin, 2023-02-08:
我也遇到了相同的问题，SSH进后台后，主题改成bootstrap也没用

d1216407601, 2023-02-08:
怎么取消，现在一直在载入视图

griphin, 2023-02-08:
取消不了，重刷默认的固件，不要订制

zengboin, 2023-02-08:
是ui.js有问题

0HUA0, 2023-02-08:
同问题，求教怎么解决？

mzfffff, 2023-02-08:
一个一个试了下是这个的问题
luci-base

0HUA0, 2023-02-08:
临时解决办法
进入 https://op.supes.top/packages
找到自己对应的处理器类型
然后进入old
搜索 luci-base
下载最近的版本
文件重命名为luci-base.ipk传到/tmp  
然后
cd /tmp
然后
opkg install luci-base.ipk -force-downgrade

mzfffff, 2023-02-08:
出问题的版本
luci-base (git-24.038.49291-ace49ef)

zengbo, 2023-02-09:
通过 SSH 登录到路由器，然后修改文件 `/www/luci-static/resources/ui.js` 第 24 行，在最后添加一个 ) 就可以了。
我把我改好的放上来，解压后，拷贝到路由器的  `/www/luci-static/resources/ui.js` 就行了。
[ui.tar.gz](https://github.com/kiddin9/OpenWrt_x86-r2s-r4s-r5s-N1/files/10691592/ui.tar.gz)

mayaxcn, 2023-02-09:
> 通过 SSH 登录到路由器，然后修改文件 `/www/luci-static/resources/ui.js` 第 24 行，在最后添加一个 ) 就可以了。 我把我改好的放上来，解压后，拷贝到路由器的 `/www/luci-static/resources/ui.js` 就行了。 [ui.tar.gz](https://github.com/kiddin9/OpenWrt_x86-r2s-r4s-r5s-N1/files/10691592/ui.tar.gz)

这方法可行，已经能进入后台了，本来还以为要重刷系统了