kiddin9/Kwrt #1237: 2月13日更新的node v16.19.0-1版本会导致unblockneteasemusic不停重启，无法启动。降级node恢复正常

tudoubocai, 2023-02-25:


dawn-lc, 2023-05-21:
```shell
Error relocating /usr/bin/node: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE9push_backEc: symbol not found
Error relocating /usr/bin/node: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE4swapERS4_: symbol not found
Error relocating /usr/bin/node: _ZNSt7__cxx1118basic_stringstreamIcSt11char_traitsIcESaIcEEC1ERKNS_12basic_stringIcS2_S3_EESt13_Ios_Openmode: symbol not found
Error relocating /usr/bin/node: _ZNKSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE4findEPKcm: symbol not found
Error relocating /usr/bin/node: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEC1ERKS4_: symbol not found
Error relocating /usr/bin/node: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE6insertEmRKS4_: symbol not found
Error relocating /usr/bin/node: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE7replaceEmmPKcm: symbol not found
Error relocating /usr/bin/node: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE6appendEPKcm: symbol not found
Error relocating /usr/bin/node: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE6resizeEm: symbol not found
Error relocating /usr/bin/node: _ZNSt7__cxx1118basic_stringstreamIcSt11char_traitsIcESaIcEEC1Ev: symbol not found
Error relocating /usr/bin/node: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE6insertEmPKc: symbol not found
Error relocating /usr/bin/node: _ZNSt7__cxx1119basic_istringstreamIcSt11char_traitsIcESaIcEEC1ERKNS_12basic_stringIcS2_S3_EESt13_Ios_Openmode: symbol not found
Error relocating /usr/bin/node: _ZNKSt7__cxx1115basic_stringbufIcSt11char_traitsIcESaIcEE3strEv: symbol not found
Error relocating /usr/bin/node: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE14_M_replace_auxEmmmc: symbol not found
Error relocating /usr/bin/node: _ZNSt7__cxx1119basic_istringstreamIcSt11char_traitsIcESaIcEED1Ev: symbol not found
Error relocating /usr/bin/node: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE12_M_constructEmc: symbol not found
Error relocating /usr/bin/node: _ZNKSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE5rfindEPKcm: symbol not found
Error relocating /usr/bin/node: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE9_M_createERmm: symbol not found
Error relocating /usr/bin/node: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE2atEm: symbol not found
Error relocating /usr/bin/node: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEC1EOS4_: symbol not found
Error relocating /usr/bin/node: _ZNSt7__cxx1119basic_ostringstreamIcSt11char_traitsIcESaIcEEC1Ev: symbol not found
Error relocating /usr/bin/node: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE7reserveEv: symbol not found
Error relocating /usr/bin/node: _ZNSt7__cxx1119basic_ostringstreamIcSt11char_traitsIcESaIcEEC1ERKNS_12basic_stringIcS2_S3_EESt13_Ios_Openmode: symbol not found
Error relocating /usr/bin/node: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE9_M_assignERKS4_: symbol not found
Error relocating /usr/bin/node: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE13_S_copy_charsEPcPKcS7_: symbol not found
Error relocating /usr/bin/node: _ZNKSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE7compareERKS4_: symbol not found
Error relocating /usr/bin/node: _ZSt7getlineIcSt11char_traitsIcESaIcEERSt13basic_istreamIT_T0_ES7_RNSt7__cxx1112basic_stringIS4_S5_T1_EES4_: symbol not found
Error relocating /usr/bin/node: _ZNKSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE8capacityEv: symbol not found
Error relocating /usr/bin/node: _ZNKSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE5rfindEcm: symbol not found
Error relocating /usr/bin/node: _ZStlsIcSt11char_traitsIcESaIcEERSt13basic_ostreamIT_T0_ES7_RKNSt7__cxx1112basic_stringIS4_S5_T1_EE: symbol not found
Error relocating /usr/bin/node: _ZStrsIcSt11char_traitsIcESaIcEERSt13basic_istreamIT_T0_ES7_RNSt7__cxx1112basic_stringIS4_S5_T1_EE: symbol not found
Error relocating /usr/bin/node: _ZNKSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE4copyEPcmm: symbol not found
Error relocating /usr/bin/node: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE5eraseEN9__gnu_cxx17__normal_iteratorIPKcS4_EES9_: symbol not found
Error relocating /usr/bin/node: _ZNSt7__cxx1118basic_stringstreamIcSt11char_traitsIcESaIcEED1Ev: symbol not found
Error relocating /usr/bin/node: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE10_M_replaceEmmPKcm: symbol not found
Error relocating /usr/bin/node: _ZNSt7__cxx1119basic_ostringstreamIcSt11char_traitsIcESaIcEEC2Ev: symbol not found
Error relocating /usr/bin/node: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE8pop_backEv: symbol not found
Error relocating /usr/bin/node: _ZNKSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE13find_first_ofEcm: symbol not found
Error relocating /usr/bin/node: _ZNKSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE11_M_is_localEv: symbol not found
Error relocating /usr/bin/node: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE6appendEPKc: symbol not found
Error relocating /usr/bin/node: _ZNKSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE13find_first_ofEPKcm: symbol not found
Error relocating /usr/bin/node: _ZNKSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE4findERKS4_m: symbol not found
Error relocating /usr/bin/node: _ZNSt7__cxx1119basic_ostringstreamIcSt11char_traitsIcESaIcEED2Ev: symbol not found
Error relocating /usr/bin/node: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE7_S_copyEPcPKcm: symbol not found
[... cut ...]
Error relocating /usr/bin/node: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE6appendERKS4_: symbol not found
Error relocating /usr/bin/node: _ZNKSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE6substrEmm: symbol not found
Error relocating /usr/bin/node: _ZNKSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE7compareEmmRKS4_: symbol not found
Error relocating /usr/bin/node: _ZNKSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE4findEcm: symbol not found
Error relocating /usr/bin/node: _ZNSt7__cxx1115basic_stringbufIcSt11char_traitsIcESaIcEE3strERKNS_12basic_stringIcS2_S3_EE: symbol not found

```

dawn-lc, 2023-05-21:
@tudoubocai 降到哪个版本正常了?

tudoubocai, 2023-05-23:
> @tudoubocai降到哪个版本正常了?node_v14.21.2-1_x86_64.ipk
去openwrt官方源里面找。找不到给我发消息，我发给你。只有x86的哦