kiddin9/Kwrt #1243: AX6 定制固件2023/1/16内核5.15.86 的问题反馈

digiw, 2023-02-28:
情况反馈，请K大修修：
1. cpu超频模块内频率范围只能选择两档，最大频率和最小频率都是这样 
[image]

2. 登陆路由后，概览首页的网络接口，只能显示一个半双工的单网口状态，实际使用没问题，确认是显示问题，之前某一版好像有过
[image]

3. Turbo ACC网络加速设置模块，在包更新里面发现有新版，在线更新了一下，发现模块里流量分载和全锥型nat不能开，得到防火墙设置里开启相应设置，如果TURBO ACC里也开了，会导致不能启用这两功能，还会断流。  因此全锥型nat下拉菜单里的性能模式就始终没启用成功。
[image]

gaifeng8864, 2023-04-10:
你好，我从20221028版本之后也发现了这个问题，于是一直没再更新新版本，就用着20221028版本。昨天更新了art8072驱动后无线连不上了，于是重新定制了最新版固件，但是你上边描述的前两个问题依然存在，而且无线依旧无法连接。请问，你的这些问题解决了吗？

digiw, 2023-04-10:
> 你好，我从20221028版本之后也发现了这个问题，于是一直没再更新新版本，就用着20221028版本。昨天更新了ath11k-firmware-ipq8074驱动后无线连不上了，于是重新定制了最新版固件，无线依旧无法连接。上边你描述的前两个问题也依然存在。请问，你的这些问题解决了吗？

无线其实是没问题的了，如果你重新定制固件，那么到ax6的uboot模式里做全新刷新，无线就不会出错了。至于我上述几个问题，cpu超频和网口的显示那是倒车老bug，要等K大修复，而trcc加速里面，如果要开全锥型nat，那么下拉模式要选兼容模式，性能模式是无法用的，可能也要等修复。

gaifeng8864, 2023-04-10:
> > 你好，我从20221028版本之后也发现了这个问题，于是一直没再更新新版本，就用着20221028版本。昨天更新了ath11k-firmware-ipq8074驱动后无线连不上了，于是重新定制了最新版固件，无线依旧无法连接。上边你描述的前两个问题也依然存在。请问，你的这些问题解决了吗？
> 
> 无线其实是没问题的了，如果你重新定制固件，那么到ax6的uboot模式里做全新刷新，无线就不会出错了。至于我上述几个问题，cpu超频和网口的显示那是倒车老bug，要等K大修复，而trcc加速里面，如果要开全锥型nat，那么下拉模式要选兼容模式，性能模式是无法用的，可能也要等修复。

感谢回复。我今天也试了进uboot模式刷新了，也是不行。难道是uboot版本不兼容？你用的是哪个版本的uboot？

digiw, 2023-04-10:
> 感谢回复。我今天也试了进uboot模式刷新了，也是不行。难道是uboot版本不兼容？你用的是哪个版本的uboot？

uboot似乎并没什么很多版本，我大概记得就一个2.0版本吧好像是，只是定制完毕，那个factory.ubi的文件，才是uboot刷新专用的。如果刷新遇到障碍，可以尝试更换浏览器（缓存的关系），我在edge，chrome，opera上都可以顺利刷新，你可以换着试试，一般能进入uboot页面，接下来问题就不大了。

gaifeng8864, 2023-04-10:
> uboot似乎并没什么很多版本，我大概记得就一个2.0版本吧好像是，只是定制完毕，那个factory.ubi的文件，才是uboot刷新专用的。如果刷新遇到障碍，可以尝试更换浏览器（缓存的关系），我在edge，chrome，opera上都可以顺利刷新，你可以换着试试，一般能进入uboot页面，接下来问题就不大了。

哈哈，是我没表达清楚，uboot模式下固件是刷成功了的，只是无线也还是不行，我的uboot也是2.0版。用20221028版没什么问题，只要不更新最新的那个无线驱动。先用着20221028版吧，除了固件版本低点，像CPU超频，网口显示，无线连接等都没有问题。再次感谢你的热心回复。

digiw, 2023-04-10:
> 哈哈，是我没表达清楚，uboot模式下固件是刷成功了的，只是无线也还是不行，我的uboot也是2.0版。用20221028版没什么问题，只要不更新最新的那个无线驱动。先用着20221028版吧，除了固件版本低点，像CPU超频，网口显示，无线连接等都没有问题。再次感谢你的热心回复。

不客气！我定制的时候倒没有特别留意wifi驱动的事，k大的网页定制很方便，新版的wifi也正常，如果你有手动换了驱动，那可能需要考虑到兼容问题，或者就等等新版发布看。

gaifeng8864, 2023-04-10:
> 不客气！我定制的时候倒没有特别留意wifi驱动的事，k大的网页定制很方便，新版的wifi也正常，如果你有手动换了驱动，那可能需要考虑到兼容问题，或者就等等新版发布看。

行，我先用着，等新版本出来了再试试。再次感谢。

digiw, 2023-04-10:
> 行，我先用着，等新版本出来了再试试。再次感谢。

想想觉得不太对劲，因为我并没有使用ax6的wifi，而是另接了ap来连接wifi的，因此以前的版本ax6 wifi是正常的我就没多在意，你这么一说以后，我试了试在1月16日以后编译的ax6固件，确实不论5g还是2.4g都没法和ax6连接，症状是显示正在连接，然后立刻就断开！ 即便是删除openwrt里的wifi设备，重新添加并设置，也还是这样，看起来是新版的固件当中wifi驱动有些问题了！

gaifeng8864, 2023-04-11:
> 想想觉得不太对劲，因为我并没有使用ax6的wifi，而是另接了ap来连接wifi的，因此以前的版本ax6 wifi是正常的我就没多在意，你这么一说以后，我试了试在1月16日以后编译的ax6固件，确实不论5g还是2.4g都没法和ax6连接，症状是显示正在连接，然后立刻就断开！ 即便是删除openwrt里的wifi设备，重新添加并设置，也还是这样，看起来是新版的固件当中wifi驱动有些问题了！

哈哈，你的测试过程和我一摸一样。看来新版驱动确实有问题了。希望能尽快修复吧。另外，之前那两个倒车的BUG已经很久了，也是没修复。是不是AX6被放弃了？

digiw, 2023-04-11:
> 哈哈，你的测试过程和我一摸一样。看来新版驱动确实有问题了。希望能尽快修复吧。另外，之前那两个倒车的BUG已经很久了，也是没修复。是不是AX6被放弃了？

我觉得还能抢救一下，在k大出手前，可以先试试自救下，
如果你方便的话，可以尝试ssh登陆到路由器，到路径/lib/modules/5.15.94/ 下，下载wifi的驱动程序（两个.ko后缀）文件，分别是：ath11k.ko 和 ath11k_ahb.ko， 
另外wifi的固件文件是在路径：/lib/firmware/ath11k/IPQ8074/hw2.0/   下，一共也是两个bin文件：board-2.bin 和 cal-ahb-c000000.wifi.bin，
让后丢个网盘我来下载替换到我新版编译的ax6固件里试试，要是可行，你要可以刷新版然后替换一下文件来搞定。

gaifeng8864, 2023-04-11:
> > > > > > > > > > > > 你好，我从20221028版本之后也发现了这个问题，于是一直没再更新新版本，就用着20221028版本。昨天更新了ath11k-firmware-ipq8074驱动后无线连不上了，于是重新定制了最新版固件，无线依旧无法连接。上边你描述的前两个问题也依然存在。请问，你的这些问题解决了吗？
> > > > > > > > > > > 
> > > > > > > > > > > 
> > > > > > > > > > > 无线其实是没问题的了，如果你重新定制固件，那么到ax6的uboot模式里做全新刷新，无线就不会出错了。至于我上述几个问题，cpu超频和网口的显示那是倒车老bug，要等K大修复，而trcc加速里面，如果要开全锥型nat，那么下拉模式要选兼容模式，性能模式是无法用的，可能也要等修复。
> > > > > > > > > > 
> > > > > > > > > > 
> > > > > > > > > > 感谢回复。我今天也试了进uboot模式刷新了，也是不行。难道是uboot版本不兼容？你用的是哪个版本的uboot？
> > > > > > > > > 
> > > > > > > > > 
> > > > > > > > > uboot似乎并没什么很多版本，我大概记得就一个2.0版本吧好像是，只是定制完毕，那个factory.ubi的文件，才是uboot刷新专用的。如果刷新遇到障碍，可以尝试更换浏览器（缓存的关系），我在edge，chrome，opera上都可以顺利刷新，你可以换着试试，一般能进入uboot页面，接下来问题就不大了。
> > > > > > > > 
> > > > > > > > 
> > > > > > > > 哈哈，是我没表达清楚，uboot模式下固件是刷成功了的，只是无线也还是不行，我的uboot也是2.0版。用20221028版没什么问题，只要不更新最新的那个无线驱动。先用着20221028版吧，除了固件版本低点，像CPU超频，网口显示，无线连接等都没有问题。再次感谢你的热心回复。
> > > > > > > 
> > > > > > > 
> > > > > > > 不客气！我定制的时候倒没有特别留意wifi驱动的事，k大的网页定制很方便，新版的wifi也正常，如果你有手动换了驱动，那可能需要考虑到兼容问题，或者就等等新版发布看。
> > > > > > 
> > > > > > 
> > > > > > 行，我先用着，等新版本出来了再试试。再次感谢。
> > > > > 
> > > > > 
> > > > > 想想觉得不太对劲，因为我并没有使用ax6的wifi，而是另接了ap来连接wifi的，因此以前的版本ax6 wifi是正常的我就没多在意，你这么一说以后，我试了试在1月16日以后编译的ax6固件，确实不论5g还是2.4g都没法和ax6连接，症状是显示正在连接，然后立刻就断开！ 即便是删除openwrt里的wifi设备，重新添加并设置，也还是这样，看起来是新版的固件当中wifi驱动有些问题了！
> > > > 
> > > > 
> > > > 哈哈，你的测试过程和我一摸一样。看来新版驱动确实有问题了。希望能尽快修复吧。另外，之前那两个倒车的BUG已经很久了，也是没修复。是不是AX6被放弃了？
> > > 
> > > 
> > > 我觉得还能抢救一下，在k大出手前，可以先试试自救下， 如果你方便的话，可以尝试ssh登陆到路由器，到路径/lib/modules/5.15.94/ 下，下载wifi的驱动程序（两个.ko后缀）文件，分别是：ath11k.ko 和 ath11k_ahb.ko， 另外wifi的固件文件是在路径：/lib/firmware/ath11k/IPQ8074/hw2.0/ 下，一共也是两个bin文件：board-2.bin 和 cal-ahb-c000000.wifi.bin， 让后丢个网盘我来下载替换到我新版编译的ax6固件里试试，要是可行，你要可以刷新版然后替换一下文件来搞定。
> > 
> > 
> > 从路由器里现在这几个文件倒是没问题，但是我目前在用20221028版的固件。新版之前不能用wifi后我就换老版固件了。按你的方法，我从新版路由器系统文件里里下载的wifi相关文件不也是有问题的吗？用它们再编译固件可行吗？
> 
> 我的意思是，你可以先备份当前20221028版固件里的这四个文件， 然后直接刷新最新编译的固件，这时候wifi肯定就不正常了对吧，那么用之前备份的文件，去新固件的相同路径替换同名文件后，重启一下路由看是否ok。
> 
> 由于不知道是.ko的驱动有问题（大概率是），还是.bin的wifi固件文件有问题，因此我建议是先替换那两个.ko的驱动文件。
> 
> 这方法其实就是用正常状态的wifi 驱动，去替换新固件里不正常的wifi驱动，替换前你可以把新编译的固件里这4个文件也备份一下。因为万一新编译固件里，如果有一头文件（譬如wifi固件.bin）并没有故障，那起码可以新旧搭配跑的舒畅些。要是怕麻烦，那就四个文件一起替换。
> 
> 这方法比较暴力些，主要是群主的定制固件似乎没有办法去选择wifi驱动这种版本定制项目。

这个方案你之前测试过吗？据说linux内核对驱动版本很敏感，不知道版本跨度这么大能不能行。

digiw, 2023-04-11:
> 这个方案你之前测试过吗？据说linux内核对驱动版本很敏感，不知道版本跨度这么大能不能行。

据分析是有可行性的，并且翻车也只是wifi不能用，由于老文件有备份，可以倒车回来也不用担心！

gaifeng8864, 2023-04-11:
> 据分析是有可行性的，并且翻车也只是wifi不能用，由于老文件有备份，可以倒车回来也不用担心！

那基本wifi问题有着落了。但是没有CPU超频和网口显示bug，强迫症不能忍啊，哈哈。

digiw, 2023-04-11:
> 那基本wifi问题有着落了。但是没有CPU超频和网口显示bug，强迫症不能忍啊，哈哈。

这个暂时靠催眠大法！实际使用中发现超频上限似乎还没法修改，应该是编译过程里写进文件里去的，这个而要等k大了，至于网口，反正眼不见为净，也不会老去看它拉，还是能忍一下的。 wifi这块期待听到你的好消息！

1299887963, 2023-04-11:
请问WiFi的问题上边的方法可以吗？如果可以，麻烦发下这几个文件，刚刚更新发现WiFi连不上了。😂😂

digiw, 2023-04-11:
> 请问WiFi的问题上边的方法可以吗？如果可以，麻烦发下这几个文件，刚刚更新发现WiFi连不上了。😂😂也没备份，果然睡觉前刷机是大忌。。。

我觉得需要@gaifeng8864  这位兄弟帮忙了，因为我和你一样，也是新版固件并且很久之前刷新ax6时，就没有用它的wifi，以至于现在才发现这是个坑，哈哈！

gaifeng8864, 2023-04-12:
> > 请问WiFi的问题上边的方法可以吗？如果可以，麻烦发下这几个文件，刚刚更新发现WiFi连不上了。😂😂也没备份，果然睡觉前刷机是大忌。。。
> 
> 我觉得需要@gaifeng8864 这位兄弟帮忙了，因为我和你一样，也是新版固件并且很久之前刷新ax6时，就没有用它的wifi，以至于现在才发现这是个坑，哈哈！
[wifi_20221028.zip](https://github.com/kiddin9/OpenWrt_x86-r2s-r4s-r5s-N1/files/11205880/wifi_20221028.zip)
 我把你说的那几个文件都整下来了，你有时间的话试试编译固件或直接更换wifi文件试试。对了，里边还有个文件ath.ko，我不知道有没有用，直接全整下来了。哈哈，咱们正好一个在跑新版固件，一个在跑旧版固件，正好接力完成这个测试，下一步就靠你了。

gaifeng8864, 2023-04-12:
> 感谢 @gaifeng8864 兄弟提供的文件，其中ath.ko 是 Atheros 网络设备的驱动程序，包括 WiFi 和 Ethernet 等，应该和用高通芯片的ax6没什么关系。 我下载了你分享的文件后做了如下测试：
> 
> 1. 仅替换.bin后缀的 wifi固件文件，重启后发现wifi无法激活，看起来openwrt内核版本必须要和wifi固件版本适配才行！
> 2. 我恢复了.bin文件，然后仅替换.ko 的wifi驱动程序文件，重启路由后发现依然不行，提示缺少依赖的模块，
> 3. 重新到“软件包” 菜单卸载了ath11相关的包，重启，再次安装好这些包重启，发现故障照旧，那么说明并不是wifi驱动程序的问题，而是最新的wifi固件和内核之间的相关问题。
>    这一点，如果刷新了最新编译版本的固件，在系统日志：内核日志的末尾，会发现大量相关wifi固件的报错信息。
> 
> 因此我建议你先不要刷新版编译的固件了，我再开个贴向作者反馈下，只能期待他在编译的源头掐掉bug了！

@digiw 兄弟辛苦了。感谢你的测试并给出这么详尽的结论和建议。那就等作者修复吧，再次感谢。

1299887963, 2023-04-12:
@gaifeng8864 @digiw 
1. 单独替换wifi固件(board-2.bin是一样的，cal-ahb-c000000.wifi.bin不同)，wifi都扫不到了；
2. wifi的驱动(ath11k.ko 和 ath11k_ahb.ko)替换后可以扫到，但一连就断开，和未替换时现象一致；
3. 同时替换wifi固件和wifi的驱动，wifi扫不到，路由器wifi管理界面显示无线未关联。
看样子替换不行。。。

digiw, 2023-04-12:
> @gaifeng8864 @digiw
> 
> 1. 单独替换wifi固件(board-2.bin是一样的，cal-ahb-c000000.wifi.bin不同)，wifi都扫不到了；
> 2. wifi的驱动(ath11k.ko 和 ath11k_ahb.ko)替换后可以扫到，但一连就断开，和未替换时现象一致；
> 3. 同时替换wifi固件和wifi的驱动，wifi扫不到，路由器wifi管理界面显示无线未关联。
>    看样子替换不行。。。

是的，已经另开贴单独反馈了。

digiw, 2023-04-16:
@gaifeng8864 @1299887963  两位，今天作者发布的最新定制版本以及修复了wifi问题，可以放心更新，比较建议用uboot刷入。至于另外几个小 bug，还得等他后续出手：））

gaifeng8864, 2023-04-16:
> @gaifeng8864 @1299887963 两位，今天作者发布的最新定制版本以及修复了wifi问题，可以放心更新，比较建议用uboot刷入。至于另外几个小 bug，还得等他后续出手：））

感谢，我看驱动好像还没有单独放出来。我想只更新驱动试试的。

digiw, 2023-04-16:
> > @gaifeng8864 @1299887963 两位，今天作者发布的最新定制版本以及修复了wifi问题，可以放心更新，比较建议用uboot刷入。至于另外几个小 bug，还得等他后续出手：））
> 
> 感谢，我看驱动好像还没有单独放出来。我想只更新驱动试试的。就是这个包ath11k-firmware-ipq8074，目前还是2023-03-31-a039049a-1版本，之前就是更新了这个版本出的问题。
[image]
你可以和图片里的版本做个比对，我在更新到新版，wifi恢复正常后，查到按照的版本如图，看起来似乎是新内核与新驱动搭配正确。

gaifeng8864, 2023-04-16:
> > > @gaifeng8864 @1299887963 两位，今天作者发布的最新定制版本以及修复了wifi问题，可以放心更新，比较建议用uboot刷入。至于另外几个小 bug，还得等他后续出手：））
> > 
> > 
> > 感谢，我看驱动好像还没有单独放出来。我想只更新驱动试试的。就是这个包ath11k-firmware-ipq8074，目前还是2023-03-31-a039049a-1版本，之前就是更新了这个版本出的问题。
> > [image]
> > 你可以和图片里的版本做个比对，我在更新到新版，wifi恢复正常后，查到按照的版本如图，看起来似乎是新内核与新驱动搭配正确。
[image]
这是我的相关文件目前的版本。目前综合看起来应该是内核兼容问题。你刷的最新版固件目前还有哪些bug？

digiw, 2023-04-16:
> @gaifeng8864 @1299887963 两位，今天作者发布的最新定制版本以及修复了wifi问题，可以放心更新，比较建议用uboot刷入。至于另外几个小 bug，还得等他后续出手：）） [...]

以我目前能感知的，就是本帖顶楼那3个问题了。

gaifeng8864, 2023-04-16:
> 以我目前能感知的，就是本帖顶楼那3个问题了。

希望今年能解决这几个问题。虽然不算大问题，但是很影响使用体验。

digiw, 2023-04-16:
> 希望今年能解决这几个问题。虽然不算大问题，但是很影响使用体验。

愿作者发力，一定会越来越好的！实际应用中，目前暂时影响不大！

gaifeng8864, 2023-04-17:
> 愿作者发力，一定会越来越好的！实际应用中，目前暂时影响不大！

是的，这款固件使用上总体还是很稳定，性能也很好，希望越来越好。实际使用中除了那个CPU超频略微影响一些响应速度外，其他确实没有什么影响，就是观感上不舒服。

digiw, 2023-05-30:
@gaifeng8864 @1299887963 两位，5.15.112新版里，亲测wifi已经修复，可以冲了

gaifeng8864, 2023-05-31:
> @gaifeng8864 @1299887963 两位，5.15.112新版里，亲测wifi已经修复，可以冲了

感谢，感谢提醒，这不就奥里给了吗，哈哈。对了，概览里的网口显示问题和CPU超频问题修复了吗？正想升级，但是担心没修复还得再折腾回去。

digiw, 2023-05-31:
> > @gaifeng8864 @1299887963 两位，5.15.112新版里，亲测wifi已经修复，可以冲了
> 
> 感谢，感谢提醒，这不就奥里给了吗，哈哈。对了，概览里的网口显示问题和CPU超频问题修复了吗？正想升级，但是担心没修复还得再折腾回去。

我感觉网口显示应该是修复了（主要我做聚合，它显示不了聚合状态，正常普通连接我感觉没问题），cpu超频这个倒还是老样子没修复。其他的，似乎没有问题了。

digiw, 2023-06-03:
> > > > @gaifeng8864 @1299887963 两位，5.15.112新版里，亲测wifi已经修复，可以冲了
> > > 
> > > 
> > > 感谢，感谢提醒，这不就奥里给了吗，哈哈。对了，概览里的网口显示问题和CPU超频问题修复了吗？正想升级，但是担心没修复还得再折腾回去。
> > 
> > 
> > 我感觉网口显示应该是修复了（主要我做聚合，它显示不了聚合状态，正常普通连接我感觉没问题），cpu超频这个倒还是老样子没修复。其他的，似乎没有问题了。
> > CPU超频这个问题看来真是个问题了。你的概览里的温度都到多少？我的室温20度多点，WIFI温度居然能到55-60度。
wifi我这里没有显示温度，cpu在53度左右，还好

gaifeng8864, 2023-06-03:
> > > > > @gaifeng8864 @1299887963 两位，5.15.112新版里，亲测wifi已经修复，可以冲了
> > > > 
> > > > 
> > > > 感谢，感谢提醒，这不就奥里给了吗，哈哈。对了，概览里的网口显示问题和CPU超频问题修复了吗？正想升级，但是担心没修复还得再折腾回去。
> > > 
> > > 
> > > 我感觉网口显示应该是修复了（主要我做聚合，它显示不了聚合状态，正常普通连接我感觉没问题），cpu超频这个倒还是老样子没修复。其他的，似乎没有问题了。
> > > CPU超频这个问题看来真是个问题了。你的概览里的温度都到多少？我的室温20度多点，WIFI温度居然能到55-60度。
> > > wifi我这里没有显示温度，cpu在53度左右，还好

那应该是我这个老版本还显示wifi温度，新版不显示了。我的CPU温度和你的差不太多，不知wifi温度为什么那么高。

digiw, 2023-06-07:
> 那应该是我这个老版本还显示wifi温度，新版不显示了。我的CPU温度和你的差不太多，不知wifi温度为什么那么高。

不好意思，我要修正一下！我uboot全新更新后，可以显示wifi温度，室温26左右时，我只开了2.4g（47度），所以感觉你的硬件似乎需要看下散热是不是可以改善，当然，建议还是uboot全新更新一下固件。

digiw, 2023-06-07:
> > > > > > > > @gaifeng8864 @1299887963 两位，5.15.112新版里，亲测wifi已经修复，可以冲了
> > > > > > > 
> > > > > > > 
> > > > > > > 感谢，感谢提醒，这不就奥里给了吗，哈哈。对了，概览里的网口显示问题和CPU超频问题修复了吗？正想升级，但是担心没修复还得再折腾回去。
> > > > > > 
> > > > > > 
> > > > > > 我感觉网口显示应该是修复了（主要我做聚合，它显示不了聚合状态，正常普通连接我感觉没问题），cpu超频这个倒还是老样子没修复。其他的，似乎没有问题了。
> > > > > > CPU超频这个问题看来真是个问题了。你的概览里的温度都到多少？我的室温20度多点，WIFI温度居然能到55-60度。
> > > > > > wifi我这里没有显示温度，cpu在53度左右，还好
> > > 
> > > 
> > > 那应该是我这个老版本还显示wifi温度，新版不显示了。我的CPU温度和你的差不太多，不知wifi温度为什么那么高。
> > 
> > 
> > 不好意思，我要修正一下！我uboot全新更新后，可以显示wifi温度，室温26左右时，我只开了2.4g（47度），所以感觉你的硬件似乎需要看下散热是不是可以改善，当然，建议还是uboot全新更新一下固件。
> 
> 看来确实需要改善一下散热。目前的最新版概览里的网口显示问题和CPU超频问题解决了吗？

网口显示正常，只有cpu超频没扩展，不过没有感觉瓶颈。

gaifeng8864, 2023-06-08:
> 网口显示正常，只有cpu超频没扩展，不过没有感觉瓶颈。

那可以上车了，感谢老兄。