kiddin9/Kwrt #1274: 源地址失效了？

icezxf, 2023-03-22:
opkg update失败 在线定制地址也进不去了

tudoubocai, 2023-03-24:
同……望尽快修复

ciel-zijie, 2023-05-02:
相同的问题

`[root@OpenWrt:12:35 PM ~] # opkg update
Downloading https://istore.linkease.com/repo/all/compat/Packages.gz
Updated list of available packages in /var/opkg-lists/istore_compat
Downloading /Packages.gz
*** Failed to download the package list from /Packages.gz

Downloading https://mirrors.aliyun.com/openwrt/packages/x86_64/base/Packages.gz
*** Failed to download the package list from https://mirrors.aliyun.com/openwrt/packages/x86_64/base/Packages.gz

Downloading https://mirrors.aliyun.com/openwrt/packages/x86_64/packages/Packages.gz
*** Failed to download the package list from https://mirrors.aliyun.com/openwrt/packages/x86_64/packages/Packages.gz

Downloading https://mirrors.aliyun.com/openwrt/packages/x86_64/luci/Packages.gz
*** Failed to download the package list from https://mirrors.aliyun.com/openwrt/packages/x86_64/luci/Packages.gz

Downloading https://mirrors.aliyun.com/openwrt/packages/x86_64/routing/Packages.gz
*** Failed to download the package list from https://mirrors.aliyun.com/openwrt/packages/x86_64/routing/Packages.gz

Downloading https://mirrors.aliyun.com/openwrt/packages/x86_64/kiddin9/Packages.gz
*** Failed to download the package list from https://mirrors.aliyun.com/openwrt/packages/x86_64/kiddin9/Packages.gz

Collected errors:
 * opkg_conf_parse_file: /etc/opkg/distfeeds.conf:2: Ignoring invalid line: `ttps://mirrors.aliyun.com/openwrt/releases/x86/64'
 * opkg_download: 下载失败 /Packages.gz, wget returned 1.
 * opkg_download: 下载失败 https://mirrors.aliyun.com/openwrt/packages/x86_64/base/Packages.gz, wget returned 8.
 * opkg_download: 下载失败 https://mirrors.aliyun.com/openwrt/packages/x86_64/packages/Packages.gz, wget returned 8.
 * opkg_download: 下载失败 https://mirrors.aliyun.com/openwrt/packages/x86_64/luci/Packages.gz, wget returned 8.
 * opkg_download: 下载失败 https://mirrors.aliyun.com/openwrt/packages/x86_64/routing/Packages.gz, wget returned 8.
 * opkg_download: 下载失败 https://mirrors.aliyun.com/openwrt/packages/x86_64/kiddin9/Packages.gz, wget returned 8.`