kiddin9/Kwrt #1278: x86全锥型nat等配置无法保存，重启消失

song3108song, 2023-03-24:
发现防火墙中全锥型nat、软件流量分载、硬件流量分载不能保存，一旦重启配置丢失；同时不能上网，需要重新点选上述选项后才可以上网。还发现bbr和全锥型nat不能同时开启。上述问题发生在现在能下载的最新版本上。

masx200, 2023-03-25:
我也发现有问题了,
/etc/rc.local
安装了turboacc之后要开机后重启服务就行
```
# Put your custom commands here that should be executed once
# the system init finished. By default this file does nothing.
sleep 60

service firewall restart

service  turboacc restart

service odhcpd restart
exit 0

```

masx200, 2023-03-26:
[image]

这个样子是不是说明硬件转发失效了？变成软件转发了？

chmdir, 2023-05-23:
> 发现防火墙中全锥型nat、软件流量分载、硬件流量分载不能保存，一旦重启配置丢失；同时不能上网，需要重新点选上述选项后才可以上网。还发现bbr和全锥型nat不能同时开启。上述问题发生在现在能下载的最新版本上。

我小米3G和小米AX6S 最新版本，都是这个问题，全锥型NAT 重启就没了，上不了网每次重启要手动打勾，还有AX6S的5GWIFI，也用不了，只能用2.4G
[image]