kiddin9/Kwrt #1284: 网站崩了？https://www.itdog.cn/http/扫一圈发现只有港澳台有响应，切港台全局代理访问的也是网页快照

lyhisso666, 2023-03-27:


lyhisso666, 2023-03-27:
<html>
<body>

检测点 | 响应IP | IP位置 | 状态 | 总耗时 | 解析 | 连接 | SSL握手 | 重定向 | Head
-- | -- | -- | -- | -- | -- | -- | -- | -- | --
电信 甘肃兰州 | 43.154.199.12 | 中国/香港/腾讯云 | 失败 | 10s | 0.018s | 失败 | -- | -- | --
电信 湖北武汉 | 43.154.199.12 | 中国/香港/腾讯云 | 失败 | 10s | 0.021s | 失败 | -- | -- | --
电信 福建泉州 | 43.154.199.12 | 中国/香港/腾讯云 | 失败 | 10.001s | 0.034s | 失败 | -- | -- | --
电信 新疆乌鲁木齐 | 43.154.199.12 | 中国/香港/腾讯云 | 失败 | 10.001s | 0.041s | 失败 | -- | -- | --
电信 湖南长沙 | 43.154.199.12 | 中国/香港/腾讯云 | 失败 | 10.001s | 0.022s | 失败 | -- | -- | --
电信 贵州贵阳 | 43.154.199.12 | 中国/香港/腾讯云 | 失败 | 10.002s | 0.015s | 失败 | -- | -- | --
电信 内蒙古 | 43.154.199.12 | 中国/香港/腾讯云 | 失败 | 10.001s | 0.046s | 失败 | -- | -- | --
电信 云南昆明 | 43.154.199.12 | 中国/香港/腾讯云 | 失败 | 10s | 0.020s | 失败 | -- | -- | --
电信 广西南宁 | 43.154.199.12 | 中国/香港/腾讯云 | 失败 | 10.001s | 0.021s | 失败 | -- | -- | --
电信 河南洛阳 | 43.154.199.12 | 中国/香港/腾讯云 | 失败 | 10.002s | 0.020s | 失败 | -- | -- | --
电信 山东青岛 | 43.154.199.12 | 中国/香港/腾讯云 | 失败 | 10.002s | 0.005s | 失败 | -- | -- | --
电信 天津 | 43.154.199.12 | 中国/香港/腾讯云 | 失败 | 10s | 0.008s | 失败 | -- | -- | --
电信 辽宁大连 | 43.154.199.12 | 中国/香港/腾讯云 | 失败 | 10s | 0.020s | 失败 | -- | -- | --
电信 山西太原 | 43.154.199.12 | 中国/香港/腾讯云 | 失败 | 10.001s | 0.018s | 失败 | -- | -- | --
电信 西藏拉萨 | 43.154.199.12 | 中国/香港/腾讯云 | 失败 | 10.001s | 0.027s | 失败 | -- | -- | --
电信 宁夏银川 | 43.154.199.12 | 中国/香港/腾讯云 | 失败 | 10.002s | 0.037s | 失败 | -- | -- | --
电信 浙江杭州 | 43.154.199.12 | 中国/香港/腾讯云 | 失败 | 10.001s | 0.007s | 失败 | -- | -- | --
电信 河北石家庄 | 43.154.199.12 | 中国/香港/腾讯云 | 失败 | 10.001s | 0.012s | 失败 | -- | -- | --
电信 浙江宁波 | 43.154.199.12 | 中国/香港/腾讯云 | 失败 | 10.001s | 0.014s | 失败 | -- | -- | --
电信 安徽滁州 | 43.154.199.12 | 中国/香港/腾讯云 | 失败 | 10.001s | 0.014s | 失败 | -- | -- | --
电信 四川成都 | 43.154.199.12 | 中国/香港/腾讯云 | 失败 | 10.002s | 0.006s | 失败 | -- | -- | --
电信 青海西宁 | 43.154.199.12 | 中国/香港/腾讯云 | 失败 | 10s | 0.024s | 失败 | -- | -- | --
电信 黑龙江哈尔滨 | 43.154.199.12 | 中国/香港/腾讯云 | 失败 | 10.002s | 0.028s | 失败 | -- | -- | --
电信 海南海口 | 43.154.199.12 | 中国/香港/腾讯云 | 失败 | 10.002s | 0.019s | 失败 | -- | -- | --
电信 江苏镇江 | 43.154.199.12 | 中国/香港/腾讯云 | 失败 | 10.001s | 0.013s | 失败 | -- | -- | --
电信 四川绵阳 | 43.154.199.12 | 中国/香港/腾讯云 | 失败 | 10.001s | 0.009s | 失败 | -- | -- | --
电信 北京 | 43.154.199.12 | 中国/香港/腾讯云 | 失败 | 10.002s | 0.004s | 失败 | -- | -- | --
电信 广东深圳 | 43.154.199.12 | 中国/香港/腾讯云 | 失败 | 10.001s | 0.008s | 失败 | -- | -- | --
电信 陕西西安 | 43.154.199.12 | 中国/香港/腾讯云 | 失败 | 10.001s | 0.022s | 失败 | -- | -- | --
电信 重庆 | 43.154.199.12 | 中国/香港/腾讯云 | 失败 | 10.002s | 0.012s | 失败 | -- | -- | --
电信 上海 | 43.154.199.12 | 中国/香港/腾讯云 | 失败 | 10s | 0.007s | 失败 | -- | -- | --
电信 江西南昌 | 43.154.199.12 | 中国/香港/腾讯云 | 失败 | 10.001s | 0.032s | 失败 | -- | -- | --
电信 福建龙岩 | 43.154.199.12 | 中国/香港/腾讯云 | 失败 | 10.001s | 0.025s | 失败 | -- | -- | --
电信 吉林长春 | 43.154.199.12 | 中国/香港/腾讯云 | 失败 | 10s | 0.029s | 失败 | -- | -- | --
联通 湖南湘潭 | 43.154.199.12 | 中国/香港/腾讯云 | 失败 | 10.002s | 0.072s | 失败 | -- | -- | --
联通 山东济南 | 43.154.199.12 | 中国/香港/腾讯云 | 失败 | 10.002s | 0.009s | 失败 | -- | -- | --
联通 广西柳州 | 43.154.199.12 | 中国/香港/腾讯云 | 失败 | 10.001s | 0.039s | 失败 | -- | -- | --
联通 贵州贵阳 | 43.154.199.12 | 中国/香港/腾讯云 | 失败 | 10s | 0.041s | 失败 | -- | -- | --
联通 辽宁大连 | 43.154.199.12 | 中国/香港/腾讯云 | 失败 | 10.002s | 0.024s | 失败 | -- | -- | --
联通 安徽芜湖 | 43.154.199.12 | 中国/香港/腾讯云 | 失败 | 10.001s | 0.023s | 失败 | -- | -- | --
联通 陕西西安 | 43.154.199.12 | 中国/香港/腾讯云 | 失败 | 10.001s | 0.040s | 失败 | -- | -- | --
联通 湖北武汉 | 43.154.199.12 | 中国/香港/腾讯云 | 失败 | 10.001s | 0.023s | 失败 | -- | -- | --
联通 河北石家庄 | 43.154.199.12 | 中国/香港/腾讯云 | 失败 | 10.001s | 0.020s | 失败 | -- | -- | --
联通 北京 | 43.154.199.12 | 中国/香港/腾讯云 | 失败 | 10s | 0.005s | 失败 | -- | -- | --
联通 新疆乌鲁木齐 | 43.154.199.12 | 中国/香港/腾讯云 | 失败 | 10.001s | 0.061s | 失败 | -- | -- | --
联通 河南郑州 | 43.154.199.12 | 中国/香港/腾讯云 | 失败 | 10.002s | 0.020s | 失败 | -- | -- | --
联通 甘肃兰州 | 43.154.199.12 | 中国/香港/腾讯云 | 失败 | 10.002s | 0.027s | 失败 | -- | -- | --
联通 云南昆明 | 43.154.199.12 | 中国/香港/腾讯云 | 失败 | 10.001s | 0.049s | 失败 | -- | -- | --
联通 黑龙江佳木斯 | 43.154.199.12 | 中国/香港/腾讯云 | 失败 | 10.001s | 0.030s | 失败 | -- | -- | --
联通 福建泉州 | 43.154.199.12 | 中国/香港/腾讯云 | 失败 | 10.002s | 0.039s | 失败 | -- | -- | --
联通 广东潮州 | 43.154.199.12 | 中国/香港/腾讯云 | 失败 | 10.002s | 0.045s | 失败 | -- | -- | --
联通 河北邯郸 | 43.154.199.12 | 中国/香港/腾讯云 | 失败 | 10.002s | 0.019s | 失败 | -- | -- | --
联通 贵州贵阳 | 43.154.199.12 | 中国/香港/腾讯云 | 失败 | 10.001s | 0.073s | 失败 | -- | -- | --
联通 江西南昌 | 43.154.199.12 | 中国/香港/腾讯云 | 失败 | 10.002s | 0.028s | 失败 | -- | -- | --
联通 重庆 | 43.154.199.12 | 中国/香港/腾讯云 | 失败 | 10.001s | 0.037s | 失败 | -- | -- | --
联通 江苏泰州 | 43.154.199.12 | 中国/香港/腾讯云 | 失败 | 10.002s | 0.035s | 失败 | -- | -- | --
联通 宁夏银川 | 43.154.199.12 | 中国/香港/腾讯云 | 失败 | 10.002s | 0.106s | 失败 | -- | -- | --
联通 吉林长春 | 43.154.199.12 | 中国/香港/腾讯云 | 失败 | 10.002s | 0.069s | 失败 | -- | -- | --
联通 海南海口 | 43.154.199.12 | 中国/香港/腾讯云 | 失败 | 10.001s | 0.145s | 失败 | -- | -- | --
联通 福建宁德 | 43.154.199.12 | 中国/香港/腾讯云 | 失败 | 10.001s | 0.042s | 失败 | -- | -- | --
联通 天津 | 43.154.199.12 | 中国/香港/腾讯云 | 失败 | 10.001s | 0.055s | 失败 | -- | -- | --
联通 青海西宁 | 43.154.199.12 | 中国/香港/腾讯云 | 失败 | 10.002s | 0.069s | 失败 | -- | -- | --
联通 四川成都 | 43.154.199.12 | 中国/香港/腾讯云 | 失败 | 10.001s | 0.004s | 失败 | -- | -- | --
联通 西藏拉萨 | 43.154.199.12 | 中国/香港/腾讯云 | 失败 | 10.001s | 0.039s | 失败 | -- | -- | --
联通 山西太原 | 43.154.199.12 | 中国/香港/腾讯云 | 失败 | 10.001s | 0.011s | 失败 | -- | -- | --
[... 133 lines cut ...]
移动 四川成都	43.154.199.12	中国/香港/腾讯云	失败	10.002s	0.006s	失败	--	--	--
移动 广东深圳	43.154.199.12	中国/香港/腾讯云	失败	10.001s	0.011s	失败	--	--	--
移动 山东济南	43.154.199.12	中国/香港/腾讯云	失败	10.002s	0.010s	失败	--	--	--
[... cut ...]
海外 新加坡	104.21.62.131	Anycast/cloudflare.com	403	0.347s	0.009s	0.199s	0.527s	--	[查看](javascript:;)
海外 中国香港	172.67.136.42	Anycast/cloudflare.com	403	0.333s	0.400s	0.199s	0.516s	--	[查看](javascript:;)
海外 马来西亚	104.21.62.131	Anycast/cloudflare.com	403	0.291s	0.041s	0.007s	0.273s	--	[查看](javascript:;)
海外 中国台湾	172.67.136.42	Anycast/cloudflare.com	403	0.606s	0.329s	0.132s	0.462s	--	[查看](javascript:;)
海外 日本东京	172.67.136.42	Anycast/cloudflare.com	403	0.252s	0.062s	0.001s	0.244s	--	[查看](javascript:;)
海外 柬埔寨	104.21.62.131	Anycast/cloudflare.com	403	0.478s	2.059s	0.077s	0.391s	--	[查看](javascript:;)

Zhengshihangcn, 2023-03-27:
我也是，XD什么网络？我是移动的

lyhisso666, 2023-03-27:
> 我也是，XD什么网络？我是移动的

电信校园网

110120787, 2023-03-28:
打不开了，啥情况？

0-0qimingzizhennan0-0, 2023-03-28:
确实打不开了

lyhisso666, 2023-03-28:
昨天我在编译的时候崩了，下午3点进去好了，没想到现在又崩了

0-0qimingzizhennan0-0, 2023-03-28:
> 昨天我在编译的时候崩了，下午3点进去好了，没想到现在又崩了

今天一上午我都没进去

lyhisso666, 2023-03-28:
我昨天崩的时候在自定义shell脚本填进去了chatgpt给我写的自动登录配置脚本然后编译的时候炸了，这会不会是网站没做SQL注入防护，shell脚本写离谱点就会炸

110120787, 2023-03-28:
昨晚编译的R4S的rom，没法pppoe拨号了，只能用以前的

gpsligit, 2023-03-30:
定制化在线更新也用不了，还以为是我固件出问题了