kiddin9/Kwrt #1292: 玩客云刷openwrt后上传速度受限 [附修复方案]

k3bc999, 2023-04-01:
[image]
如图所示，当使用玩客云v1.3的版型时，部分板子会出现上传速度极低，以及不停重传的现象
这种原因是因为网卡时钟问题（3.10内核时不会出现这种问题）
具体可以参考这里https://github.com/hzyitc/armbian-onecloud/issues/2
根据armbian里的解决方案，需要对dtb进行些许修改
修改后的dtb已加于附件（由群友蓝蓝进行修改）
[op-5.10.xx-fix-uplink-dtb.zip](https://github.com/kiddin9/OpenWrt_x86-r2s-r4s-r5s-N1/files/11128413/op-5.10.xx-fix-uplink-dtb.zip)
将其解压改名为dtb覆盖于1分区即可
修复后效果如下图所示
[image]

希望大佬能进行修复 便于后续在线编译

另外还想提个请求，就是lede源码的大佬说，主线uboot不支持玩客云，然后他把emmc驱动去掉了，但是Hzyitc（上面那个armbian）的ubooot支持，因此希望这里的openwrt也能支持一下，这样就可以做线刷包了，也不需要snail的底包了

十分感谢

Haugor, 2023-04-07:
感谢大佬分享，折腾了一周，反复设置刷固件一直没搞定。。。十分感谢

kiddin9, 2023-04-07:
感谢提供解决方案,

k3bc999, 2023-04-08:
> 感谢提供解决方案,

作者大大辛苦了

cs0663c, 2023-05-26:
我的玩客云刷完内网能跑800左右， 用来做单臂路由拨号，测速最大也就80m 百兆都跑不满，不知道什么问题 ，用usb网口做wan 跑40多m 百兆宽带都跑不了

lanlanpy, 2023-05-27:
> 我的玩客云刷完内网能跑800左右， 用来做单臂路由拨号，测速最大也就80m 百兆都跑不满，不知道什么问题 ，用usb网口做wan 跑40多m 百兆宽带都跑不了
> 
> 

单口复用，本来就会有损耗的，加上玩客云目前内核还有dts针对网卡部分不完善