kiddin9/Kwrt #1323: 用的新路由3（NEWIFI3 D2）和 京东云鲁班  wifi

644234041, 2023-04-18:


644234041, 2023-04-18:
用的新路由3（NEWIFI3 D2）和 京东云鲁班  wifi都有问题
手机、电脑任意设备wifi连接路由器
每个几秒钟 或者10几秒钟 就会出现高延迟的情况 或者访问失败。
无意中发现手机刷抖音 卡住刷不出来新的视频  
后来去ping了才发现这个问题（2个路由器 不同移动设备 测试都有这个问题）
来自 192.168.100.1 的回复: 字节=32 时间=1ms TTL=64
来自 192.168.100.1 的回复: 字节=32 时间=2673ms TTL=64
来自 192.168.100.1 的回复: 字节=32 时间=5ms TTL=64
来自 192.168.100.1 的回复: 字节=32 时间=2ms TTL=64
来自 192.168.100.1 的回复: 字节=32 时间=3ms TTL=64
来自 192.168.100.1 的回复: 字节=32 时间=1ms TTL=64
来自 192.168.100.1 的回复: 字节=32 时间=4ms TTL=64
来自 192.168.100.1 的回复: 字节=32 时间=3ms TTL=64
来自 192.168.100.1 的回复: 字节=32 时间=2ms TTL=64
来自 192.168.100.1 的回复: 字节=32 时间=2ms TTL=64
来自 192.168.100.1 的回复: 字节=32 时间=2ms TTL=64
来自 192.168.100.1 的回复: 字节=32 时间=3ms TTL=64
来自 192.168.100.1 的回复: 字节=32 时间=2ms TTL=64
来自 192.168.100.1 的回复: 字节=32 时间=2ms TTL=64
来自 192.168.100.1 的回复: 字节=32 时间=2657ms TTL=64
来自 192.168.100.1 的回复: 字节=32 时间=3ms TTL=64
来自 192.168.100.1 的回复: 字节=32 时间=1ms TTL=64
来自 192.168.100.1 的回复: 字节=32 时间=3ms TTL=64
来自 192.168.100.1 的回复: 字节=32 时间=3ms TTL=64
来自 192.168.100.1 的回复: 字节=32 时间=2ms TTL=64
来自 192.168.100.1 的回复: 字节=32 时间=4ms TTL=64
来自 192.168.100.1 的回复: 字节=32 时间=3ms TTL=64
来自 192.168.100.1 的回复: 字节=32 时间=2ms TTL=64
来自 192.168.100.1 的回复: 字节=32 时间=2ms TTL=64
来自 192.168.100.1 的回复: 字节=32 时间=2ms TTL=64
来自 192.168.100.1 的回复: 字节=32 时间=5ms TTL=64
来自 192.168.100.1 的回复: 字节=32 时间=2ms TTL=64
来自 192.168.100.1 的回复: 字节=32 时间=3ms TTL=64
来自 192.168.100.1 的回复: 字节=32 时间=4ms TTL=64
来自 192.168.100.1 的回复: 字节=32 时间=1ms TTL=64
来自 192.168.100.1 的回复: 字节=32 时间=2673ms TTL=64
来自 192.168.100.1 的回复: 字节=32 时间=5ms TTL=64
来自 192.168.100.1 的回复: 字节=32 时间=2ms TTL=64
来自 192.168.100.1 的回复: 字节=32 时间=3ms TTL=64
来自 192.168.100.1 的回复: 字节=32 时间=1ms TTL=64
来自 192.168.100.1 的回复: 字节=32 时间=1ms TTL=64
来自 192.168.100.1 的回复: 字节=32 时间=2ms TTL=64
来自 192.168.100.1 的回复: 字节=32 时间=2ms TTL=64
[image]

yuanshus, 2023-05-20:
> 

你的newifi d2用的是什么版本的内核