kiddin9/Kwrt #1328: x86更新到04-19版本后wifi无法启动

FdolaLily, 2023-04-19:
无线安全处显示：WPA 加密需要安装 wpa_supplicant（客户端模式）或安装 hostapd（接入点 AP、点对点 Ad-Hoc 模式）。
使用opkg安装 hostapd 显示 
Not downgrading package hostapd-common on root from 2022-07-29-b704dc72-0.2 to 2022-01-16-cff80b4f-16.2.
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency hostapd-common (= 2022-07-29-b704dc72-24.2) for hostapd-basic-mbedtls
 * pkg_hash_check_unresolved: cannot find dependency hostapd-common (= 2022-07-29-b704dc72-24.2) for hostapd-mbedtls
 * pkg_hash_check_unresolved: cannot find dependency hostapd-common (= 2022-07-29-b704dc72-24.2) for wpad-basic-mbedtls
 * pkg_hash_check_unresolved: cannot find dependency hostapd-common (= 2022-07-29-b704dc72-24.2) for wpad-mbedtls
 * pkg_hash_check_unresolved: cannot find dependency hostapd-common (= 2022-07-29-b704dc72-24.2) for wpad-mesh-mbedtls
 * opkg_install_cmd: Cannot install package hostapd.

FdolaLily, 2023-04-19:
无线用的英特尔 ax210

9thChasingWindGirl, 2023-04-19:
Xiaomi Mi Router AC2100设备5Gwifi连接异常（IPV6通信正常），多次断开并手动重连后，5G WIFI显示没有IPV6地址，但可以正常使用。

9thChasingWindGirl, 2023-04-19:
> Xiaomi Mi Router AC2100设备5Gwifi连接异常（IPV6通信正常），多次断开并手动重连后，5G WIFI显示没有IPV6地址，但可以正常使用。

进一步观察后，发现两个频段都不能获取到IPV6地址，但是2.4G频段没有获取到IPV6地址仍然能通过V4联网，5G频段则不行