kiddin9/Kwrt #1371: 红米ax6 wifi不稳是什么问题 连上就断了

lianyanjiajia, 2023-05-14:


Ai-desu-2333, 2023-05-14:
5.15.109版本内核，红米AX6，WiFi同样出现此情况；

通过观察表现为无线设备反复开启和关闭，下面是相关日志

<details>
<summary>系统日志：</summary>
<pre><code>
涉及MAC的隐私内容替换为了[MAC]
Mon May 15 01:19:09 2023 daemon.warn netifd: Wireless device 'radio0' setup failed, retry=3
Mon May 15 01:19:09 2023 daemon.notice netifd: radio1 (17559): Command failed: Not found
Mon May 15 01:19:09 2023 daemon.notice netifd: radio0 (17560): Command failed: Not found
Mon May 15 01:19:09 2023 daemon.notice ttyd[6894]: rops_handle_POLLIN_netlink: DELADDR
Mon May 15 01:19:09 2023 kern.info kernel: [  189.591672] br-lan: port 3(phy1-ap0) entered disabled state
Mon May 15 01:19:10 2023 kern.info kernel: [  189.729290] br-lan: port 4(phy0-ap0) entered disabled state
Mon May 15 01:19:10 2023 kern.info kernel: [  189.779479] device phy1-ap0 left promiscuous mode
Mon May 15 01:19:10 2023 kern.info kernel: [  189.779507] br-lan: port 3(phy1-ap0) entered disabled state
Mon May 15 01:19:10 2023 kern.info kernel: [  189.859574] device phy0-ap0 left promiscuous mode
Mon May 15 01:19:10 2023 kern.info kernel: [  189.859602] br-lan: port 4(phy0-ap0) entered disabled state
Mon May 15 01:19:10 2023 daemon.notice netifd: Wireless device 'radio1' is now down
Mon May 15 01:19:10 2023 daemon.notice netifd: Wireless device 'radio0' is now down
Mon May 15 01:19:10 2023 daemon.notice netifd: radio1 (17586): sh: acs_survey: out of range
Mon May 15 01:19:10 2023 daemon.notice netifd: radio1 (17586): sh: acs_survey: out of range
Mon May 15 01:19:10 2023 daemon.notice hostapd: Configuration file: /var/run/hostapd-phy1.conf (phy phy1-ap0) --> new PHY
Mon May 15 01:19:10 2023 daemon.notice hostapd: phy1-ap0: interface state UNINITIALIZED->COUNTRY_UPDATE
Mon May 15 01:19:10 2023 daemon.notice hostapd: ACS: Automatic channel selection started, this may take a bit
Mon May 15 01:19:10 2023 daemon.notice hostapd: phy1-ap0: interface state COUNTRY_UPDATE->ACS
Mon May 15 01:19:10 2023 daemon.notice hostapd: phy1-ap0: ACS-STARTED
Mon May 15 01:19:10 2023 kern.info kernel: [  190.312335] br-lan: port 3(phy1-ap0) entered blocking state
Mon May 15 01:19:10 2023 kern.info kernel: [  190.312365] br-lan: port 3(phy1-ap0) entered disabled state
Mon May 15 01:19:10 2023 kern.info kernel: [  190.312725] device phy1-ap0 entered promiscuous mode
Mon May 15 01:19:10 2023 daemon.notice hostapd: Configuration file: /var/run/hostapd-phy0.conf (phy phy0-ap0) --> new PHY
Mon May 15 01:19:10 2023 daemon.notice hostapd: phy0-ap0: interface state UNINITIALIZED->COUNTRY_UPDATE
Mon May 15 01:19:10 2023 daemon.notice hostapd: ACS: Automatic channel selection started, this may take a bit
Mon May 15 01:19:10 2023 daemon.notice hostapd: phy0-ap0: interface state COUNTRY_UPDATE->ACS
Mon May 15 01:19:10 2023 daemon.notice hostapd: phy0-ap0: ACS-STARTED
Mon May 15 01:19:10 2023 kern.info kernel: [  190.343761] br-lan: port 4(phy0-ap0) entered blocking state
Mon May 15 01:19:10 2023 kern.info kernel: [  190.343788] br-lan: port 4(phy0-ap0) entered disabled state
Mon May 15 01:19:10 2023 kern.info kernel: [  190.344075] device phy0-ap0 entered promiscuous mode
Mon May 15 01:19:10 2023 kern.info kernel: [  190.344218] br-lan: port 4(phy0-ap0) entered blocking state
Mon May 15 01:19:10 2023 kern.info kernel: [  190.344232] br-lan: port 4(phy0-ap0) entered forwarding state
Mon May 15 01:19:11 2023 kern.info kernel: [  191.349113] br-lan: port 4(phy0-ap0) entered disabled state
Mon May 15 01:19:11 2023 daemon.notice netifd: Wireless device 'radio1' is now up
Mon May 15 01:19:11 2023 daemon.notice netifd: Wireless device 'radio0' is now up
Mon May 15 01:19:19 2023 daemon.notice hostapd: phy1-ap0: ACS-COMPLETED freq=2412 channel=1
Mon May 15 01:19:19 2023 kern.info kernel: [  199.508265] IPv6: ADDRCONF(NETDEV_CHANGE): phy1-ap0: link becomes ready
Mon May 15 01:19:19 2023 kern.info kernel: [  199.508544] br-lan: port 3(phy1-ap0) entered blocking state
Mon May 15 01:19:19 2023 kern.info kernel: [  199.508569] br-lan: port 3(phy1-ap0) entered forwarding state
Mon May 15 01:19:19 2023 daemon.notice netifd: Network device 'phy1-ap0' link is up
Mon May 15 01:19:19 2023 daemon.notice hostapd: phy1-ap0: interface state ACS->ENABLED
Mon May 15 01:19:19 2023 daemon.notice hostapd: phy1-ap0: AP-ENABLED
Mon May 15 01:19:22 2023 daemon.info hostapd: phy1-ap0: STA [MAC] IEEE 802.11: authenticated
Mon May 15 01:19:22 2023 daemon.info hostapd: phy1-ap0: STA [MAC] IEEE 802.11: associated (aid 1)
Mon May 15 01:19:22 2023 daemon.notice hostapd: phy1-ap0: AP-STA-CONNECTED [MAC] auth_alg=open
Mon May 15 01:19:22 2023 daemon.info hostapd: phy1-ap0: STA [MAC] WPA: pairwise key handshake completed (RSN)
Mon May 15 01:19:22 2023 daemon.notice hostapd: phy1-ap0: EAPOL-4WAY-HS-COMPLETED [MAC]
Mon May 15 01:19:24 2023 daemon.info hostapd: phy1-ap0: STA [MAC] IEEE 802.11: authenticated
Mon May 15 01:19:24 2023 daemon.info hostapd: phy1-ap0: STA [MAC] IEEE 802.11: associated (aid 2)
Mon May 15 01:19:25 2023 daemon.notice hostapd: phy1-ap0: AP-STA-CONNECTED [MAC] auth_alg=open
Mon May 15 01:19:25 2023 daemon.info hostapd: phy1-ap0: STA [MAC] WPA: pairwise key handshake completed (RSN)
Mon May 15 01:19:25 2023 daemon.notice hostapd: phy1-ap0: EAPOL-4WAY-HS-COMPLETED [MAC]
Mon May 15 01:19:30 2023 daemon.notice netifd: Network device 'phy1-ap0' link is down
Mon May 15 01:19:30 2023 kern.info kernel: [  210.288022] br-lan: port 3(phy1-ap0) entered disabled state
Mon May 15 01:19:30 2023 daemon.warn netifd: Wireless device 'radio1' setup failed, retry=3
[... cut ...]
[   85.317553] br-lan: port 4(phy0-ap0) entered blocking state
[   85.317579] br-lan: port 4(phy0-ap0) entered disabled state
[   85.317932] device phy0-ap0 entered promiscuous mode
[   85.318095] br-lan: port 4(phy0-ap0) entered blocking state
[   85.318113] br-lan: port 4(phy0-ap0) entered forwarding state
[   86.311876] br-lan: port 4(phy0-ap0) entered disabled state
[   94.431675] IPv6: ADDRCONF(NETDEV_CHANGE): phy1-ap0: link becomes ready
</code></pre>
</details>

退回到本地保留的旧版本固件 内核版本5.15.94，WiFi恢复正常。

harry0071, 2023-05-15:
5.15.109的固件有问题，wifi频繁断连
我回退到5.15.1.06就好了

digiw, 2023-05-30:
5.15.112 新版已经修复了